OpenIntelWireless/itlwm #101: Question WPA2-Enterprise Support

KristFlex, 2020-05-31:
Hello! 
First, thank you so much for your effort! You are doing such a great job. 
I just wanted to know if the kext supports WPA2-Enterprise (802.1X) Is it planned to support it in the future or is it already supported? I remember that all the kexts for realtek usb wifi never supported this standard. 
Thank you in advance

Suzamax, 2020-06-16:
Some resource I found about openbsd's driver for 802.1X http://undeadly.org/cgi?action=article&sid=20130128142215

Also, PEAP/MSCHAP support might be nice: maybe porting iked(8) would work, isn't it? I just requested this in another repo.

usr-sse2, 2020-08-06:
@KristFlex Actually Realtek USB WiFi drivers (that with status bar app, not a windowed utility) support WPA2-Enterprise. They create a configuration file and run `wpa_supplicant`. However, if the certificate is not trusted, `wpa_supplicant` crashes while trying to present a window to approve or reject a certificate, but I created a config and run `wpa_supplicant` manually and it works.

zxystd, 2020-11-10:
Considering of the certificate of WPA2 Enterprise network, I don't want to implement this in itlwm.kext, if anyone has this requirement, please use AirportItlwm.kext instead.

closed 2020-11-10 by zxystd

xchgre, 2020-11-19:
It would be awesome a stable .kext with wpa2_enterprise support, is safe to use AirportItlwm.kext and supported¿

diwu20, 2020-12-07:
It seems that WPA2 Enterprise network is still not supported in newest version AirportItlwm.kext.

HsunGong, 2021-01-28:
Still not supported

Johnson-Lsx, 2021-09-02:
I an now using the latest stable version of airportitlwm, when I try to connect to my campus wifi, which uses a wap2_enterprise certificate, it asks me to text into the username and password again and again, but can not connect to it.

zxystd, 2021-09-02:
@Johnson-Lsx which card? Product Id.

foolely, 2022-02-07:
It seems that WPA2 Enterprise network is still not supported in newest version AirportItlwm.kext. [v2.1.0 stable] it cannot connect successfully.

williambj1, 2022-02-09:
@foolely Based on feedback this is working for quite a lot of people, would you mind sharing logs with us to help identify the issue?

foolely, 2022-02-21:
> 

how to get the logs for feedback? @williambj1

AbhiMayadam, 2022-04-13:
> > 
> 
> how to get the logs for feedback? @williambj1

https://openintelwireless.github.io/itlwm/Troubleshooting.html#debugging-802-1x
It is listed here and uses the default macos wifi diagnostic utility.
Here is my log when attempting to connect to an enterprise network. I am on the latest stable Airportitlwm.kext and I use an Intel AC 7265 RevD wifi card. 
https://files.catbox.moe/k4qo5i.gz

AbhiMayadam, 2022-04-13:
I used catbox.moe for the file as I couldnt upload it to github directly. It just got renamed for some reason

Frostyman758, 2026-02-15:
> > [@usr-sse2](https://github.com/usr-sse2) will WPA2-Enterprise support be implemented?
> 
> It is implemented, I use it every day at work. Though it is rather tricky to connect, needs many tries, sometimes disabling and enabling Wi-Fi or going into sleep and back helps.

in aeroport or itlwm? because I dont think heliport can do WPA2 so I will need to kil sip