OpenIntelWireless/itlwm #98: 昨天用了半天后断网了

okexi, 2020-05-31:
我是放到OC(.58)下加载的，系统是15.5，网卡8265AC，机型小米Pro，连接的是5G频段，从下午一点到晚上10点用的都没有问题，22：20 左右网断了，当时我在bili看视频（看了有10分钟了），在此之前下了一部电影，网速可以，能保持在4M/S左右，大约在22：23分左右，我执行了
`log show --last 10m|grep itlwm >>log.log`
这个命令，得到了如下的结果，
[log.log](https://github.com/zxystd/itlwm/files/4706921/log.log)
我也不清楚是不是个例，希望能对开发者有所帮助

okexi, 2020-05-31:
用的是昨天你在远景发布的编译好的驱动

zxystd, 2020-05-31:
you are doing heavy upload i think,and the queue was full.

it disconnected in few time but it can reconnect automatically alright?

okexi, 2020-05-31:
> you are doing heavy upload i think,and the queue was full.
> 
> it disconnected in few time but it can reconnect automatically alright?

过了五分钟左右没恢复我就关机了，今天再试试，不是BUG就好

zxystd, 2020-05-31:
额，算是bug也可以不算，目前高负载对它来说还是有点勉强了，多观察一下吧，出问题的时候帮忙留意看看有没有什么规律。

okexi, 2020-05-31:
> 额，算是bug也可以不算，目前高负载对它来说还是有点勉强了，多观察一下吧，出问题的时候帮忙留意看看有没有什么规律。

好  我今天尝试复现一下

hesed, 2020-06-01:
和楼主一样的机器配置，小米pro，15.5，EFI v1.3.9（https://github.com/daliansky/XiaoMi-Pro-Hackintosh/releases），intel蓝牙连接鼠标，itlwm连接家用5Gwifi
1、finder下连samba，复制大文件到本机，一般复制到2-3G，wifi连接就会崩溃断开，不能上网，此时如果尝试启动其他应用程序，会发现速度异常缓慢。
2、正常浏览网页，时间长了，也会发生相同情况，wifi崩溃断开不能上网。

zxystd, 2020-06-03:
用最新版的试试?

closed 2020-06-13 by zxystd