OpenIntelWireless/itlwm #99: Invalid signature -67030

shomerchang, 2020-05-31:
驱动加载到最后 总是提示
Invalid signature -67030 for kext <OSKext 0x7fc5bcf25970 [0x7fff84a16b60]> { URL = "file:///Users/zhang/Desktop/itlwm.kext/", ID = "com.zxystd.itlwm" }
不知道为什么
我是ac9560

Basstorm, 2020-05-31:
Successfully loaded /Users/zhang/Desktop/itlwm.kext.
这不是成功了么
下面的可以忽略

shomerchang, 2020-05-31:
> Successfully loaded /Users/zhang/Desktop/itlwm.kext.
> 这不是成功了么
> 下面的可以忽略

但是网络里 什么变化也没有

zxystd, 2020-05-31:
log show --last boot | grep itlwm >itlwm.txt

看看日志

armen181, 2020-05-31:
I receive almost the same message. OS 10.15.4, nothing happens  
[itlwm.txt](https://github.com/zxystd/itlwm/files/4707573/itlwm.txt)

shomerchang, 2020-05-31:
[itlwm.txt](https://github.com/zxystd/itlwm/files/4707643/itlwm.txt)
这是我的日志 @zxystd

zxystd, 2020-05-31:
productid是不是02f0?如果是的话这是二代设备，等itlwmx可以用吧

shomerchang, 2020-05-31:
> productid是不是02f0?如果是的话这是二代设备，等itlwmx可以用吧

[image]
看不懂这图

zxystd, 2020-05-31:
a370是可以驱动的，彻底断电关机几分钟，再启动试试

williambj1, 2020-05-31:
https://github.com/zxystd/itlwm/issues/31#issuecomment-617542780

shomerchang, 2020-05-31:
> [#31 (comment)](https://github.com/zxystd/itlwm/issues/31#issuecomment-617542780)

niubility,成功了

shomerchang, 2020-05-31:
> a370是可以驱动的，彻底断电关机几分钟，再启动试试

清了一下 那些配置文件，成功了，感谢大佬，大佬加油！

closed 2020-05-31 by shomerchang