OpenIntelWireless/itlwm #102: Intel 8260 appears as Ethernet but can't connect.

hillfieldhonest, 2020-06-01:
Hi, 首先很感谢 这个IntelWIfi Project！我最近买到 Intel Compute Stick(m3) (which comes with Intel Wifi 8260) 本来USB端口也不多，所以我也一直找个办法如何用内置的IntelWifi.  and finally I found this awesome project! hopefully I can get mine work too.  

My Env:
HW: Compute Stick STK2M3W64CC
Installation: OpenCore 0.5.8 refering to following guide (for Skylake Desktop) 
https://dortania.github.io/OpenCore-Desktop-Guide/
kext: I used the binary one from here.
https://www.tonymacx86.com/threads/success-working-intel-wifi-drivers-for-7265ac-on-catalina.292207/page-49#post-2129921

Steps:
loaded itlwm kext manually. after editting info.plist (password only includes numbers)
cp the kext to tmp. then
sudo chown R root:wheel /tmp/itlwm.kext
sudo kextutil -v 6 /tmp/itlwm.kext 

Result:
kext loaded and it will show as Ethernet but it doesn't connect to internet.

note:
log file and some screenshots attached. let me know if more files are needed.
[log.log](https://github.com/zxystd/itlwm/files/4709459/log.log)
[image]
[image]

blickers, 2020-06-01:
Be careful when editing info.plist: password comes before ssid. Check if u've filled them correctly.

hillfieldhonest, 2020-06-02:
thank you for the advise. I double checked my info.plist just in case.
I believe I wrote it in correct way. are there anywhere else to check?

[image]

zxystd, 2020-06-02:
[image]
Only 'MakotoWifi2' in the scan result, no 'MakotoWifi', check it?

hillfieldhonest, 2020-06-02:
thank you for checking. and sorry for the confusion.
I actually had changed the ssid and retested it.
here is the log after the change. I can find the correct ssid "MakotoWifi" here...
[log_2.log](https://github.com/zxystd/itlwm/files/4714877/log_2.log)

zxystd, 2020-06-02:
itlwm: sending auth to 14:dd:a9:a3:2a:9d on channel 6 mode 11g
itlwm: authentication timed out for 14:dd:a9:a3:2a:9d

Router didn't accept the auth, can you tell me the WiFi configuration? Now it can only connect 40mhz bandwidth, 11n mode WiFi.

hillfieldhonest, 2020-06-02:
I'm using Wi-Fi tethering of Android (ZenFone 2 ZE551ML according to the spec, it supports a/ac/b/g/n. but there is no setting available to set which it use for the Wifi tethering feature.)

so, currently it tries to connect 40Mhz bandwidth 11n mode Wifi? it's good to know, I can try with other routers or phones..

hillfieldhonest, 2020-06-02:
I tried connecting to the "MakotoWifi" to know what the connection type is: it shows like this:

at least it connect using 11n mode. however not sure about 40Mhz bandwidth or not though..

[image]

zxystd, 2020-06-02:
And you check password is correct? Maybe you can use another wifi to test...

hillfieldhonest, 2020-06-02:
yes. I double checked the password.
I also tried another Androids' wifi tethering, and also Windows' mobile hotspot but in vain...
I will test again with another. and will update with the result. 

btw, eventually. 20Mhz 11n mode/or other mode will be supported?

hillfieldhonest, 2020-06-02:
I tried with the last hope, i mean only one left wi-fi router. 

and It worked!!!! I changed the router setting to 802.11/a/n/ac mode (5Ghz) , and set bandwidth to auto (since there was no 40Mhz option.)  and now it works like a magic! 

I really appreciate your great help. 
I'm posting this using the Compute stick using this itlwm kext. I'm really moved. 
[image]
[image]

zxystd, 2020-06-02:
Also glad to help you.

closed 2020-06-13 by zxystd