OpenIntelWireless/itlwm #104: Upload speed is abnormal. 下载速度正常，但上传速度很慢，掉速严重，不正常。

blickers, 2020-06-01:
问题：下载满速，上传极慢。
情景：局域网内路由器上建立的FTP服务器，用有线网连接，或者用USB外置网卡连接，上传下载都能达到满速。换 itlwm 5月30日代码编译的驱动连接，下载能够满速，但上传极慢，而且速度波动很大，刚开始能够达到1M/s，很快掉速，在几百k/s内大幅波动，再过十几秒掉到 80~90多k/s，最后维持在90k/s左右。

FTP (home NAS) upload speed is very slow. u guys can test the upload speed to confirm this issue.

blickers, 2020-06-01:
Issue #80 had been closed 3 days ago, but it seems the problem still remains.

zxystd, 2020-06-01:
It is because of the TX queue full, issue #80 was fixed, because the firmware will not crash anymore. Now the driver should not work under big pressure until improved the 11n protocol implementation.

johnnync13, 2020-07-15:
11ac or 11ax Will it be implemented in the future?? @zxystd

zxystd, 2020-07-15:
> 11ac or 11ax Will it be implemented in the future?? @zxystd

yes, step by step.

closed 2020-07-15 by zxystd