OpenIntelWireless/itlwm #106: Intel Dual Band AC3165 - Test

arch-dev, 2020-06-01:
Hi, after using latest commits no more kernel panic. So thank you again @zxystd. I have open this issue to post SpeedTest and other related test for AC3165 chipset.
Here there are the best SpeedTest results for it. If I can test something else please tell me @zxystd.
[speedtests-2020-06-01-193643.csv.zip](https://github.com/zxystd/itlwm/files/4713223/speedtests-2020-06-01-193643.csv.zip)

m0d16l14n1, 2020-06-01:
@arch-dev Hello, do you use WEP or WPA2? Can you share compiled kext please? I had 3165 too, but i use some old compiled kext and it`s didn`t work with WPA2

arch-dev, 2020-06-01:
I have a mixed security protocol (WPA/WPA2) so I don't know which of those I'm using now but soon I will update prebuilt kext so you can try.

m0d16l14n1, 2020-06-01:
@arch-dev okay, waiting. big thx!

leopck, 2020-06-01:
I'm using AC3165 as well. Here's my speedtest for some comparison. MY actual internet speed is 300Mbps.

[image]

Running AC3165 on WPA/WPA2 is no issue. Besides the audio via Intel Bluetooth being affected so far, it's just the speed performance on this could be better.

arch-dev, 2020-06-02:
@m0d16l14n1 try this kext
[itlwm.kext.zip](https://github.com/zxystd/itlwm/files/4716523/itlwm.kext.zip)

m0d16l14n1, 2020-06-02:
@arch-dev works great, thx for sharing it with me! Got 32 Mbps download and 11.73 Mbps upload using WPA/WPA2. That's great. Btw, WiFi 5 is supported or not?

williambj1, 2020-06-03:
no, only Wi-Fi 4, 802.11n

m0d16l14n1, 2020-06-03:
@williambj1 okay. Even with WiFi it`s cool. I can stop use tethering. That`s great work! Big thx @zxystd!

closed 2020-06-05 by zxystd