OpenIntelWireless/itlwm #112: Intel® Centrino® Ultimate-N 6300 support

Alexander-69, 2020-06-05:
Can you get  Intel® Centrino® Ultimate-N 6300to work?
[https://ark.intel.com/content/www/us/en/ark/products/59474/intel-centrino-ultimate-n-6300-dual-band.html](https://ark.intel.com/content/www/us/en/ark/products/59474/intel-centrino-ultimate-n-6300-dual-band.html)

zxystd, 2020-06-06:
These cards are out of date, we have more better choices. So they will not be supported. Thanks.

giovannidirenzo, 2020-06-06:
That’s sad

zxystd, 2020-06-06:
OpenBSD has `iwn` driver support this card, and I can port it in itlwm, but I don't have too much time doing this(need to debug and post maintenance), there is still leave too much thing need to do in new cards support. So Sorry to say this.

closed 2020-06-08 by zxystd

zxystd, 2020-11-18:
@Alexander-69 @tardar now you can use the latest 1.2.0-alpha version to test.

althiometer, 2020-12-02:
i have a 6205 Thinkpad Carbon, do i need to compile itlwm to test it?

althiometer, 2020-12-02:
nvm, got it, gonna test now