OpenIntelWireless/itlwm #114: 崩溃重启，详见日志，并且kext放到OC目录开机不自动加载没反应

lifsux, 2020-06-06:
panic(cpu 0 caller 0xffffff800a4c8ed0): mbuf 0xffffff8128f3b900 redzone violation with value 0x24953600 (instead of 0xc6c66ed8, using cookie 0xee35d7d8)

Backtrace (CPU 0), Frame : Return Address
0xffffff8128f3b4c0 : 0xffffff8009f1f5cd 
0xffffff8128f3b510 : 0xffffff800a058b05 
0xffffff8128f3b550 : 0xffffff800a04a68e 
0xffffff8128f3b5a0 : 0xffffff8009ec5a40 
0xffffff8128f3b5c0 : 0xffffff8009f1ec97 
0xffffff8128f3b6c0 : 0xffffff8009f1f087 
0xffffff8128f3b710 : 0xffffff800a6c27cc 
0xffffff8128f3b780 : 0xffffff800a4c8ed0 
0xffffff8128f3b7a0 : 0xffffff800a498d94 
0xffffff8128f3b7b0 : 0xffffff7f8e0e4190 
0xffffff8128f3b7e0 : 0xffffff7f8e0e5ad6 
0xffffff8128f3b970 : 0xffffff7f8e08b38f 
0xffffff8128f3b9c0 : 0xffffff7f8e080b5e 
0xffffff8128f3ba80 : 0xffffff7f8e0d00a8 
0xffffff8128f3bb20 : 0xffffff7f8e0e1c1e 
0xffffff8128f3bc00 : 0xffffff7f8e0e291e 
0xffffff8128f3be10 : 0xffffff7f8e0d6e7e 
0xffffff8128f3be70 : 0xffffff7f8e0d75f7 
0xffffff8128f3bee0 : 0xffffff800a62e60d 
0xffffff8128f3bf30 : 0xffffff800a62cede 
0xffffff8128f3bf70 : 0xffffff800a62c4d6 
0xffffff8128f3bfa0 : 0xffffff8009ec513e 
      Kernel Extensions in backtrace:
         com.zxystd.itlwm(1.0)[1FFEE7DB-C2FD-3241-98EA-6A13D42F591E]@0xffffff7f8e076000->0xffffff7f8efbcfff
            dependency: com.apple.iokit.IONetworkingFamily(3.4)[8119788F-2497-3C32-8325-C8B8D843BB31]@0xffffff7f8b0c0000
            dependency: com.apple.iokit.IOPCIFamily(2.9)[BE052F4D-9B80-3FCD-B36D-BACB7DEE0DF2]@0xffffff7f8ab01000

BSD process name corresponding to current thread: kernel_task
Boot args: darkwake=0 gfxrst=1 igfxonln=1 igfxonlnfbs=0x010203

Mac OS version:
19F101

Kernel version:
Darwin Kernel Version 19.5.0: Tue May 26 20:41:44 PDT 2020; root:xnu-6153.121.2~2/RELEASE_X86_64
Kernel UUID: 54F1A78D-6F41-32BD-BFED-4381F9F6E2EF
Kernel slide:     0x0000000009c00000
Kernel text base: 0xffffff8009e00000
__HIB  text base: 0xffffff8009d00000
System model name: MacBookPro14,1 (Mac-B4831CEBD52A0C4C)
System shutdown begun: NO
Panic diags file available: YES (0x0)

System uptime in nanoseconds: 23888607667944

zxystd, 2020-06-06:
kext来源于哪里？用最新的提交代码编译一下试试。
oc确认是可以加载的，检查你的efi配置。

lifsux, 2020-06-06:
> kext来源于哪里？用最新的提交代码编译一下试试。
> 
> oc确认是可以加载的，检查你的efi配置。

在你的pcbeta帖子下载的最新的。oc其他的kext正常加载，这个拖进去没有加载，不明原因

zxystd, 2020-06-06:
是不是忘记在config里配了?
另外你的Wi-Fi加密类型是什么,wpa2/tkip?

lifsux, 2020-06-06:
> 是不是忘记在config里配了?
> 
> 另外你的Wi-Fi加密类型是什么,wpa2/tkip?

wpa/wpa2混合加密。
我是直接拖进去OC的efi的，自动加载不会配置，能指点一下吗？我试试

lifsux, 2020-06-06:
> 是不是忘记在config里配了?
> 
> 另外你的Wi-Fi加密类型是什么,wpa2/tkip?

https://github.com/johnnync13/Xiaomi-Mi-Air我用的是这个里面的OC

zxystd, 2020-06-06:
[itlwm.kext.zip](https://github.com/zxystd/itlwm/files/4739190/itlwm.kext.zip)

这是最新的提交编译的.就算之前的我用了差不多一星期了也没崩溃过.你可以试试.

oc不像clover,丢到efi里就可以加载了,还要配置才会加载,具体你可以百度一下,就是修改config.plist,把驱动信息加进去

lifsux, 2020-06-06:
> [itlwm.kext.zip](https://github.com/zxystd/itlwm/files/4739190/itlwm.kext.zip)
> 
> 
> 
> 这是最新的提交编译的.就算之前的我用了差不多一星期了也没崩溃过.你可以试试.
> 
> 
> 
> oc不像clover,丢到efi里就可以加载了,还要配置才会加载,具体你可以百度一下,就是修改config.plist,把驱动信息加进去

好的谢谢。0603版本偶尔不定时会崩溃，比之前版本概率低多了。我专门开了一晚上直播测试的，早上起来崩溃重启了。

lifsux, 2020-06-06:
> [itlwm.kext.zip](https://github.com/zxystd/itlwm/files/4739190/itlwm.kext.zip)
> 
> 
> 
> 这是最新的提交编译的.就算之前的我用了差不多一星期了也没崩溃过.你可以试试.
> 
> 
> 
> oc不像clover,丢到efi里就可以加载了,还要配置才会加载,具体你可以百度一下,就是修改config.plist,把驱动信息加进去

自动加载解决了。
问题：1.不能识别定位
2.网速较慢不稳定，不过能上就行。主要诉求还是稳定不崩溃。
加油你很厉害！

zxystd, 2020-06-06:
网速较慢指的是多少?我这边100m宽带,能跑60-80mbps的速度,看4k没问题.

lifsux, 2020-06-06:
> 网速较慢指的是多少?我这边100m宽带,能跑60-80mbps的速度,看4k没问题.

200带宽，我手机测速两百多，电脑speedtest.cn测速四五十兆，有时候不能正常测速，不过浏览网页啥的基本正常

lifsux, 2020-06-06:
> 网速较慢指的是多少?我这边100m宽带,能跑60-80mbps的速度,看4k没问题.

[image]

lifsux, 2020-06-06:
> 网速较慢指的是多少?我这边100m宽带,能跑60-80mbps的速度,看4k没问题.

[image]

> 网速较慢指的是多少?我这边100m宽带,能跑60-80mbps的速度,看4k没问题.

[image]
[image]

这是你今天编译的最新的测速结果，延迟和抖动稍微有点问题。这个带宽1/4也能满足需求，实际我是移动200M带宽。
我路由器小米AX1800 WiFi6, wpa/wpa2

lifsux, 2020-06-06:
有需要测试的工作你可以给我安排，我配合你

zxystd, 2020-06-06:
log show --last 30m | grep itlwm >~/Desktop/itlwm_log.txt

输出一份30分钟的日志看一下,如果文件太大就缩小10分钟.不应该只有32mbps.

lifsux, 2020-06-06:
> log show --last 30m | grep itlwm >~/Desktop/itlwm_log.txt
> 
> 输出一份30分钟的日志看一下,如果文件太大就缩小10分钟.不应该只有32mbps.

https://cloud.189.cn/t/Bb6vmavquUNn
https://cloud.189.cn/t/z2qQjmIJbAve
2个日志链接

lifsux, 2020-06-06:
> log show --last 30m | grep itlwm >~/Desktop/itlwm_log.txt
> 
> 输出一份30分钟的日志看一下,如果文件太大就缩小10分钟.不应该只有32mbps.

[itlwm_log.txt](https://github.com/zxystd/itlwm/files/4739724/itlwm_log.txt)
[itlwm_log1.txt](https://github.com/zxystd/itlwm/files/4739727/itlwm_log1.txt)

zxystd, 2020-06-08:
使用最新的版本吧

lifsux, 2020-06-08:
> 使用最新的版本吧

[itlwm_log.txt](https://github.com/zxystd/itlwm/files/4745837/itlwm_log.txt)
[image]

lifsux, 2020-06-10:
> 使用最新的版本吧

错误重启：
panic(cpu 2 caller 0xffffff801f84aa3a): Kernel trap at 0xffffff801f80b580, type 14=page fault, registers:
CR0: 0x000000008001003b, CR2: 0xffffff81351cc000, CR3: 0x00000000376db000, CR4: 0x00000000003626e0
RAX: 0xffffff81351cbbd0, RBX: 0x0000000000000001, RCX: 0x0000000000000428, RDX: 0xfffffffffffffff8
RSP: 0xffffff81351cba58, RBP: 0xffffff81351cba60, RSI: 0x0000000000000006, RDI: 0x0000000000000007
R8:  0xffffff81351cbbd7, R9:  0x0000000000000007, R10: 0xffffff81351cbbd7, R11: 0xfffffffffffffffd
R12: 0x0000000000000000, R13: 0x0000000000000000, R14: 0xffffff8046019440, R15: 0xffffff810456b000
RFL: 0x0000000000010282, RIP: 0xffffff801f80b580, CS:  0x0000000000000008, SS:  0x0000000000000000
Fault CR2: 0xffffff81351cc000, Error code: 0x0000000000000002, Fault CPU: 0x2, PL: 0, VF: 10

Backtrace (CPU 2), Frame : Return Address
0xffffff81351cb4b0 : 0xffffff801f71f5cd 
0xffffff81351cb500 : 0xffffff801f858b05 
0xffffff81351cb540 : 0xffffff801f84a68e 
0xffffff81351cb590 : 0xffffff801f6c5a40 
0xffffff81351cb5b0 : 0xffffff801f71ec97 
0xffffff81351cb6b0 : 0xffffff801f71f087 
0xffffff81351cb700 : 0xffffff801fec27cc 
0xffffff81351cb770 : 0xffffff801f84aa3a 
0xffffff81351cb8f0 : 0xffffff801f84a738 
0xffffff81351cb940 : 0xffffff801f6c5a40 
0xffffff81351cb960 : 0xffffff801f80b580 
0xffffff81351cba60 : 0xffffff7fa39e26b5 
0xffffff81351cbab0 : 0xffffff7fa39e1e73 
0xffffff81351cbde0 : 0x0 
      Kernel Extensions in backtrace:
         com.zxystd.itlwm(1.0)[A2F6B5BB-09E7-3616-832E-F96758EF2E86]@0xffffff7fa39b0000->0xffffff7fa48f6fff
            dependency: com.apple.iokit.IONetworkingFamily(3.4)[8119788F-2497-3C32-8325-C8B8D843BB31]@0xffffff7fa0065000
            dependency: com.apple.iokit.IOPCIFamily(2.9)[BE052F4D-9B80-3FCD-B36D-BACB7DEE0DF2]@0xffffff7fa0131000

BSD process name corresponding to current thread: kernel_task
Boot args: darkwake=0 gfxrst=1 igfxonln=1 igfxonlnfbs=0x010203

Mac OS version:
Not yet set

Kernel version:
Darwin Kernel Version 19.5.0: Tue May 26 20:41:44 PDT 2020; root:xnu-6153.121.2~2/RELEASE_X86_64
Kernel UUID: 54F1A78D-6F41-32BD-BFED-4381F9F6E2EF
Kernel slide:     0x000000001f400000
Kernel text base: 0xffffff801f600000
__HIB  text base: 0xffffff801f500000
System model name: MacBookPro14,1 (Mac-B4831CEBD52A0C4C)
System shutdown begun: NO
Panic diags file available: YES (0x0)

System uptime in nanoseconds: 1200539610

文件：
[itlwm_log.txt](https://github.com/zxystd/itlwm/files/4755827/itlwm_log.txt)

zxystd, 2020-06-10:
没看到具体的报错。。。目前你这里还是第一个反馈崩溃的

lifsux, 2020-06-10:
> 没看到具体的报错。。。目前你这里还是第一个反馈崩溃的

错误日志在哪怎么提取给你？

zxystd, 2020-06-10:
这个已经是错误日志了，是什么情况下发生的重启？

lifsux, 2020-06-10:
> 这个已经是错误日志了，是什么情况下发生的重启？

电脑在运行但是黑屏，屏幕没有唤醒，我强制重启后显示这个崩溃错误

zxystd, 2020-06-10:
电脑睡眠了？那有可能是睡眠问题，强制重启打断了驱动的某些操作，才会报错。

lifsux, 2020-06-10:
可能是睡眠不成功，屏幕不亮了但是电脑一直没有睡眠。

发自我的iPhone

lifsux, 2020-06-10:
> 没看到具体的报错。。。目前你这里还是第一个反馈崩溃的

顺便反馈一下蓝牙，4.x设备无法连接，我的鼠标连接不了

closed 2020-06-13 by zxystd