OpenIntelWireless/itlwm #116: 热启动后断网|网速明显慢于windows下网速

lilmik, 2020-06-08:
用的是最新的6.7日的驱动，热启动之后断网。
同样测试环境下，黑苹果的intel无线网卡网速约为win10下的四分之一。
日志如下。
[itlwm_log.txt](https://github.com/zxystd/itlwm/files/4744480/itlwm_log.txt)

zxystd, 2020-06-08:
网卡型号？网卡Product ID？
网速慢是指的多少mbps

lilmik, 2020-06-08:
> 网卡型号？网卡Product ID？
> 网速慢是指的多少mbps
网卡型号：Killer Wireless-AC 1550i，网卡Product ID应该是9560吧？
网速在win10下用5g wifi 测试上传下载都能有250Mbps，在黑苹果下只有50Mbps左右 
[image]

lilmik, 2020-06-08:
mac下网络测试：
wifi状态：
[image]
有线状态：
[image]

lilmik, 2020-06-08:
Win10 下测速
wifi状态:
[image]
有线状态:
[image]

zxystd, 2020-06-08:
正常，因为现在只有11n 20mhz的速度，而且没有tx聚合，自然速度慢，慢速是60~80mbps左右吧，这不属于bug。

lilmik, 2020-06-08:
原来是这个原因，谢谢~
已经能使用上就相当不错了~
我有试了下Linux Deepin的速度，可能也是因为驱动的原因，速度还要更慢~
[image]

zxystd, 2020-06-08:
嗯，等后续开发支持更多802.11特性吧，这样速度就会上去了。

lilmik, 2020-06-08:
感谢~
话说，6.7版本的驱动比6.3版本的驱动热启动掉网卡好像更严重。。。。

zxystd, 2020-06-08:
热启动的问题我需要人跟我一起调试，具体是驱动固件版本不一致还是pcie的状态未重置导致的问题我不确定，我自己的3165,8260,9260全都没有这个问题，根本没办法测试，好像只有9560才有这个问题。

closed 2020-06-13 by zxystd