OpenIntelWireless/itlwm #117: My Wireless is 9560AC, device ID 0x9DF0 (CNVİ)

mehmetyukcell, 2020-06-08:
I'm using Catalina 10.15.5 with Clover Bootloader. After loading kext or Clover/Kext/Others, network shows Ethernet 2 (not WI-FI so there is no SSID to connect). And Ethernet 2 doesn't receive DHCP, apparently.
[image]

tvhung83, 2020-06-15:
Same here (used to work before), logs:
```
default	11:16:22.240933+0700	kernel	itlwm: iwm_match devId=0xA308
default	11:16:22.240936+0700	kernel	itlwm: free
default	11:16:22.241115+0700	kernel	itlwm: iwm_match devId=0xA323
default	11:16:22.241118+0700	kernel	itlwm: free
default	11:16:22.241312+0700	kernel	itlwm: iwm_match devId=0xA36F
default	11:16:22.241316+0700	kernel	itlwm: free
default	11:16:22.241362+0700	kernel	itlwm: iwm_match devId=0xA370
default	11:16:22.241482+0700	kernel	itlwm: iwm_match devId=0x3E9B
default	11:16:22.241485+0700	kernel	itlwm: free
default	11:16:22.241900+0700	kernel	itlwm: iwm_match devId=0xA360
default	11:16:22.241903+0700	kernel	itlwm: iwm_match devId=0x3EC2
default	11:16:22.241904+0700	kernel	itlwm: free
default	11:16:22.241905+0700	kernel	itlwm: free
default	11:16:22.241915+0700	kernel	itlwm: iwm_match devId=0xA324
default	11:16:22.241919+0700	kernel	itlwm: free
default	11:16:22.251642+0700	kernel	itlwm: iwm_attach
default	11:16:22.251801+0700	kernel	itlwm: , (null)
default	11:16:22.251821+0700	kernel	itlwm: iwm_prepare_card_hw
default	11:16:22.252360+0700	kernel	itlwm: alloc contig
default	11:16:22.346376+0700	kernel	itlwm: config ieee80211
default	11:16:22.346379+0700	kernel	itlwm: taskq taskq_create_thread lock
default	11:16:22.346384+0700	kernel	itlwm: taskq taskq_create_thread unlock
default	11:16:22.346384+0700	kernel	itlwm: taskq taskq_create_thread lock
default	11:16:22.346385+0700	kernel	ieee80211_ifattach
default	11:16:22.346386+0700	kernel	itlwm: taskq taskq_create_thread unlock
default	11:16:22.346389+0700	kernel	itlwm: taskq taskq_create_thread lock
default	11:16:22.346389+0700	kernel	itlwm: ieee80211_channel_init
default	11:16:22.346390+0700	kernel	itlwm: taskq taskq_create_thread unlock
default	11:16:22.346393+0700	kernel	itlwm: taskq taskq_thread schedule task
default	11:16:22.346393+0700	kernel	itlwm: taskq taskq_create_thread lock
default	11:16:22.346395+0700	kernel	itlwm: ieee80211_set_link_state nstate=2, old_state=-1
default	11:16:22.346395+0700	kernel	itlwm: taskq taskq_thread schedule task
default	11:16:22.346396+0700	kernel	itlwm: taskq taskq_create_thread unlock
default	11:16:22.346401+0700	kernel	itlwm: ieee80211_set_link_state LINK_STATE_IS_DOWN
default	11:16:22.346405+0700	kernel	itlwm: ieee80211_media_init
default	11:16:22.346410+0700	kernel	itlwm: iwm_preinit
default	11:16:22.346410+0700	kernel	itlwm: iwm_prepare_card_hw
default	11:16:22.346436+0700	kernel	itlwm: iwm_start_hw
default	11:16:22.346437+0700	kernel	itlwm: iwm_prepare_card_hw
default	11:16:22.351588+0700	kernel	itlwm: iwm_apm_init
default	11:16:22.351640+0700	kernel	itlwm: itlwm: L1 Disabled - LTR Enabled
default	11:16:22.352322+0700	kernel	itlwm: iwm_init_msix_hw
default	11:16:22.352324+0700	kernel	itlwm: iwm_conf_msix_hw
default	11:16:22.352371+0700	kernel	itlwm: iwm_run_init_mvm_ucode justnvm=1
default	11:16:22.352374+0700	kernel	itlwm: iwm_load_ucode_wait_alive ucode_type=1
default	11:16:22.353652+0700	kernel	itlwm: load firmware done
default	11:16:22.353662+0700	kernel	itlwm: itlwm: Paging: paging enabled (size = 241664 bytes)
default	11:16:22.353668+0700	kernel	itlwm: iwm_start_fw ucode_type=1
default	11:16:22.353671+0700	kernel	itlwm: iwm_apm_init
default	11:16:22.353718+0700	kernel	itlwm: itlwm: L1 Disabled - LTR Enabled
default	11:16:22.353951+0700	kernel	itlwm: iwm_load_firmware
default	11:16:23.354897+0700	kernel	itlwm: itlwm: fw chunk addr 0x404000 len 712 failed to load
default	11:16:23.354921+0700	kernel	itlwm: itlwm: could not load firmware chunk 0 (error 1)
default	11:16:23.354928+0700	kernel	itlwm: itlwm: failed to load init firmware
default	11:16:23.354934+0700	kernel	itlwm: iwm_stop_device
default	11:16:23.403028+0700	kernel	itlwm: iwm_conf_msix_hw
default	11:16:23.403050+0700	kernel	itlwm: iwm_prepare_card_hw
default	11:16:23.403074+0700	kernel	itlwm: attach succeed.
default	11:16:23.403088+0700	kernel	itlwm: attach to interface fail
default	11:16:23.403109+0700	kernel	         com.zxystd.itlwm(1.0)[C85237B6-FD9A-3187-A116-6E2390257BD1]@0xffffff7f85203000->0xffffff7f8617dfff
```

supdo, 2020-06-16:
Is 9560 still not driveable？

richibrics, 2020-06-16:
> I'm using Catalina 10.15.5 with Clover Bootloader. After loading kext or Clover/Kext/Others, network shows Ethernet 2 (not WI-FI so there is no SSID to connect). And Ethernet 2 doesn't receive DHCP, apparently.
> 
> [image]
> 
> 

I think you should edit ssid and password from source code, cause the interface isn’t seen as a wireless card

ErrorErrorError, 2020-06-17:
> I'm using Catalina 10.15.5 with Clover Bootloader. After loading kext or Clover/Kext/Others, network shows Ethernet 2 (not WI-FI so there is no SSID to connect). And Ethernet 2 doesn't receive DHCP, apparently.
> [image]

You're supposed to compile the kext and edit info.plist with the SSID and password you want it to connect. You can read the Readme.md docs to see why you won't see the WiFi interface.

mehmetyukcell, 2020-06-22:
Thanks for your works @zxystd I am using ac9560 and work final compiled kext with heliport.

scutxd, 2020-06-23:
> Thanks for your works @zxystd I am using ac9560 and work final compiled kext with heliport https://github.com/zxystd/HeliPort

Is ac9650 work well？

mehmetyukcell, 2020-06-23:
> > Thanks for your works @zxystd I am using ac9560 and work final compiled kext with heliport https://github.com/zxystd/HeliPort
> 
> Does ac9650 work well？My laptop is mechrevo s1 with ac9560.

Yes, 
[HeliPort.zip](https://github.com/zxystd/itlwm/files/4819650/HeliPort.zip)
[Uploading itlwm_0618.kext.zip…]()

mehmetyukcell, 2020-06-23:
kext other folder and heliport users and groups on the start

scutxd, 2020-07-17:
Great, my 9560 has worked well. But there is another issue the nic should be cold boot. If the laptop reboot in windows10, the nic will not work.

williambj1, 2020-07-17:
@supdo This issue was fixed 2 hours ago, please check the latest code

closed 2020-07-17 by williambj1

scutxd, 2020-07-18:
Thank you. i see