OpenIntelWireless/itlwm #119: 8265ac还是会不定时掉线

renyichan, 2020-06-08:
提供了日志,希望对问题有b帮助
[Uploading itlwm_log

.txt…]()

renyichan, 2020-06-08:
2020-06-08 20:53:52.791474+0800 0x7b62     Default     0x0                  3547   0    sudo:    renyi : TTY=ttys001 ; PWD=/Users/renyi ; USER=root ; COMMAND=/bin/cp -r /Users/driver/itlwm.kext /tmp
2020-06-08 20:53:52.832275+0800 0x7b67     Default     0x0                  3550   0    sudo:    renyi : TTY=ttys001 ; PWD=/Users/renyi ; USER=root ; COMMAND=/usr/libexec/PlistBuddy -c Set :IOKitPersonalities:itlwm:WiFiConfig:WiFi:ssid TP-LINK_1000 /tmp/itlwm.kext/Contents/Info.plist
2020-06-08 20:53:52.860144+0800 0x7b6d     Default     0x0                  3554   0    sudo:    renyi : TTY=ttys001 ; PWD=/Users/renyi ; USER=root ; COMMAND=/usr/libexec/PlistBuddy -c Set :IOKitPersonalities:itlwm:WiFiConfig:WiFi:password RENYI520 /tmp/itlwm.kext/Contents/Info.plist
2020-06-08 20:53:52.887197+0800 0x7b72     Default     0x0                  3557   0    sudo:    renyi : TTY=ttys001 ; PWD=/Users/renyi ; USER=root ; COMMAND=/bin/chmod -R 755 /tmp/itlwm.kext
2020-06-08 20:53:52.912928+0800 0x7b77     Default     0x0                  3560   0    sudo:    renyi : TTY=ttys001 ; PWD=/Users/renyi ; USER=root ; COMMAND=/bin/chmod 644 /tmp/itlwm.kext/Contents/Info.plist
2020-06-08 20:53:52.937861+0800 0x7b7c     Default     0x0                  3563   0    sudo:    renyi : TTY=ttys001 ; PWD=/Users/renyi ; USER=root ; COMMAND=/usr/sbin/chown -R root:wheel /tmp/itlwm.kext
2020-06-08 20:53:52.964526+0800 0x7b81     Default     0x0                  3566   0    sudo:    renyi : TTY=ttys001 ; PWD=/Users/renyi ; USER=root ; COMMAND=/usr/bin/codesign --force --deep --sign - /tmp/itlwm.kext
2020-06-08 20:53:53.005308+0800 0x7b89     Default     0x0                  3569   0    sudo:    renyi : TTY=ttys001 ; PWD=/Users/renyi ; USER=root ; COMMAND=/usr/bin/kextutil -q /tmp/itlwm.kext
2020-06-08 20:53:53.365218+0800 0x7b8b     Default     0x0                  0      0    kernel: (itlwm) _start(struct kmod_info*, void*) has been invoked
2020-06-08 20:53:53.372207+0800 0x7b98     Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_match devId=0x5916
2020-06-08 20:53:53.372217+0800 0x7b98     Default     0x0                  0      0    kernel: (itlwm) itlwm: free
2020-06-08 20:53:53.378866+0800 0x7ba3     Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_match devId=0x9D21
2020-06-08 20:53:53.378871+0800 0x7ba3     Default     0x0                  0      0    kernel: (itlwm) itlwm: free
2020-06-08 20:53:53.379426+0800 0x7b9e     Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_match devId=0x9D3A
2020-06-08 20:53:53.379431+0800 0x7b9e     Default     0x0                  0      0    kernel: (itlwm) itlwm: free
2020-06-08 20:53:53.379758+0800 0x7b99     Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_match devId=0x1911
2020-06-08 20:53:53.379766+0800 0x7b99     Default     0x0                  0      0    kernel: (itlwm) itlwm: free
2020-06-08 20:53:53.380076+0800 0x7ba6     Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_match devId=0x24FD
2020-06-08 20:53:53.380203+0800 0x7ba5     Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_match devId=0x9D23
2020-06-08 20:53:53.380222+0800 0x7ba5     Default     0x0                  0      0    kernel: (itlwm) itlwm: free
2020-06-08 20:53:53.380645+0800 0x7ba2     Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_match devId=0x9D4E
2020-06-08 20:53:53.380650+0800 0x7ba2     Default     0x0                  0      0    kernel: (itlwm) itlwm: free
2020-06-08 20:53:53.381747+0800 0x7b9b     Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_match devId=0x9D31
2020-06-08 20:53:53.381750+0800 0x7b9b     Default     0x0                  0      0    kernel: (itlwm) itlwm: free
2020-06-08 20:53:53.381836+0800 0x7b97     Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_match devId=0x5914
2020-06-08 20:53:53.381840+0800 0x7b97     Default     0x0                  0      0    kernel: (itlwm) itlwm: free
2020-06-08 20:53:53.393662+0800 0x7ba6     Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_attach
2020-06-08 20:53:53.394275+0800 0x7ba6     Default     0x0                  0      0    kernel: (itlwm) itlwm: , (null)
2020-06-08 20:53:53.394553+0800 0x7ba6     Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_prepare_card_hw
2020-06-08 20:53:53.395319+0800 0x7ba6     Default     0x0                  0      0    kernel: (itlwm) itlwm: alloc contig
2020-06-08 20:53:53.560628+0800 0x7ba6     Default     0x0                  0      0    kernel: (itlwm) itlwm: config ieee80211
2020-06-08 20:53:53.560638+0800 0x7ba6     Default     0x0                  0      0    kernel: (itlwm) ieee80211_ifattach
2020-06-08 20:53:53.560641+0800 0x7ba6     Default     0x0                  0      0    kernel: (itlwm) itlwm: ieee80211_channel_init
2020-06-08 20:53:53.560646+0800 0x7ba6     Default     0x0                  0      0    kernel: (itlwm) itlwm: ieee80211_set_link_state nstate=2, old_state=-1
2020-06-08 20:53:53.560648+0800 0x7ba6     Default     0x0                  0      0    kernel: (itlwm) itlwm: ieee80211_set_link_state LINK_STATE_IS_DOWN
2020-06-08 20:53:53.560652+0800 0x7ba6     Default     0x0                  0      0    kernel: (itlwm) itlwm: ieee80211_media_init
[... cut ...]
2020-06-08 20:59:16.484706+0800 0x7ba7     Default     0x0                  0      0    kernel: (itlwm) itlwm: ieee80211_node_choose_bss nextbs
2020-06-08 20:59:16.484711+0800 0x7ba7     Default     0x0                  0      0    kernel: (itlwm) itlwm: itlwm: - dc:2a:14:2b:e6:47
2020-06-08 20:59:16.484714+0800 0x7ba7     Default     0x0                  0      0    kernel: (itlwm) itlwm:   13
2020-06-08 20:59:16.484717+0800 0x7ba7     Default     0x0                  0      0    kernel: (itlwm) itlwm:   +20
2020-06-08 20:59:16.484719+0800 0x7ba7     Default     0x0                  0      0    kernel: (itlwm) itlwm:  5

zxystd, 2020-06-08:
应该是跟路由器有关,我看到一大堆的错误包.

zxystd, 2020-06-08:
断开了会重练吧?应该没有卡死的情况了吧

zxystd, 2020-06-13:
检查一下路由器的MTU是不是设置的1500

closed 2020-08-15 by williambj1