OpenIntelWireless/itlwm #120: Upload Kext

tolgasonkaya, 2020-06-08:
Hello there. Sometimes I can't compile kext with XCode. Upload latest kext in Google Drive please.

H15teve, 2020-06-09:
You can download the compiled kext in Chinese forum PCBeta. Visit it from README

Yomanz, 2020-06-09:
> You can download the compiled kext in Chinese forum PCBeta. Visit it from README

The forums require an invitation code apparently.

williambj1, 2020-06-09:
Learn Xcode or wait, it's up to you.

Yomanz, 2020-06-09:
I ran `xcodebuild` but it didn't give me a kext.

levisliujunqi, 2020-06-09:
[itlwm0607.kext.zip](https://github.com/zxystd/itlwm/files/4752120/itlwm0607.kext.zip)
I downloaded it for you from PCBeta

Yomanz, 2020-06-09:
> [itlwm0607.kext.zip](https://github.com/zxystd/itlwm/files/4752120/itlwm0607.kext.zip)
> I downloaded it for you from PCBeta

Thank you very much

closed 2020-06-13 by zxystd

johnnync13, 2020-06-13:
running xcode to me the kext not appear in folder. if i use fw_gen.sh it appears this:
./fw_gen.sh: line 20: /itlwm/FwBinary.cpp: No such file or directory
./fw_gen.sh: line 21: /itlwm/FwBinary.cpp: No such file or directory
./fw_gen.sh: line 35: /itlwm/FwBinary.cpp: No such file or directory
./fw_gen.sh: line 37: /itlwm/FwBinary.cpp: No such file or directory
./fw_gen.sh: line 46: /itlwm/FwBinary.cpp: No such file or directory
./fw_gen.sh: line 47: /itlwm/FwBinary.cpp: No such file or directory