OpenIntelWireless/itlwm #121: 9560 not working

Yomanz, 2020-06-09:
I installed the kext with `sudo kextutil -v 6 itlwm.kext`

I rebooted and there was nothing new in networking, I used `log show | grep itlwm` to see if there were any errors.
Here are the results.

https://ptero.co/raw/rocowuvoxa

zxystd, 2020-06-10:
This is a warm reboot issue, if you reboot from windows, you should shutdown and unplug the power cable, wait a few minute and boot, it will work. Because when windows shutdown, it leave the card a bad state.

williambj1, 2020-06-11:
@Yomanz 

It makes no sense that batteries are still supplying the entire motherboard after you shutdown your laptop, PCI cards should be powered off entirely, and that's how my ASUS laptop behaves. The card remains in a powered state when the power cable is in, and the firmware is still in it, this design is probably made to support WOW.

Instead of arguing with your "theory", why not give it a try, it only takes you 2 minutes.

mehmetyukcell, 2020-06-11:
> @Yomanz
>
>
>
> It makes no sense that batteries are still supplying the entire motherboard after you shutdown your laptop, PCI cards should be powered off entirely, and that's how my ASUS laptop [...]

How behaves 
İntel 9560 CNVi from Hp Probook 450 G6

Yomanz, 2020-06-11:
> @Yomanz
>
> It makes no sense that batteries are still supplying the entire motherboard after you shutdown your laptop, PCI cards should be powered off entirely, and that's how my ASUS laptop behaves. [...]

I followed the initial instructions, booted into windows, completely shut down the machine, left it for 5minutes, attempted to boot back into MacOS and it made no difference.

The power cable wasn't plugged in at any time and I'm unsure how else to test the theory.

Yomanz, 2020-06-11:
Reinstalling the kext fixed it.

closed 2020-06-11 by Yomanz