OpenIntelWireless/itlwm #127: AX201 Support

svandive, 2020-06-12:
Trying to load this kext to get an Intel AX 201 (Intel NUC 10) up and running.  However, after doing a kextload the drive doesn't seem to be loading.  Are there logs that would be helpful in understanding the support of this kext for AX 201?

svandive, 2020-06-14:
Fantastic, thank you!

simon-marcus, 2020-06-15:
@zxystd thank you for your development! I would also be glad to test and share feedback on the AX201 which is presently the included card in my [Gigabyte Z-490 Vision D](https://www.gigabyte.com/Motherboard/Z490-VISION-D-rev-1x/sp#sp).

svandive, 2020-06-19:
I just wanted to add a log from attempting to load the itlwmx kext.

[itlwmx_load_log.txt](https://github.com/zxystd/itlwm/files/4805434/itlwmx_load_log.txt)

purecmd, 2020-07-04:
I got a same problem. ASUS ROG STRIX Z490-I GAMING

chyinan, 2020-07-22:
小米笔记本2020增强版的AX201（AC9560）网卡也不行，什么时候能支持呢

zncs520, 2020-07-22:
cnvio2接口的目前是都不行的
我是ICE LAKE G7 哎 显卡也是没有驱动

zxystd, 2020-07-27:
ax201 is same chip with 9462, please turn to #88

closed 2020-07-27 by zxystd