OpenIntelWireless/itlwm #129: ax201不能加载

htmambo, 2020-06-12:
截取了部分日志，看着好像是buffer不够
`
2020-06-13 03:22:53.676500+0800 0x6280     Default     0x0                  0      0    kernel: (itlwmx) itlwm: : applying debug destination 1
2020-06-13 03:22:53.868313+0800 0x6280     Default     0x0                  0      0    kernel: (itlwmx) itlwm: allocDmaMemory2 alloc DMA memory failed.
2020-06-13 03:22:53.987455+0800 0x6280     Default     0x0                  0      0    kernel: (itlwmx) itlwm: allocDmaMemory2 alloc DMA memory failed.
2020-06-13 03:22:54.069948+0800 0x6280     Default     0x0                  0      0    kernel: (itlwmx) itlwm: : allocated 0x00100000 bytes for firmware monitor.
2020-06-13 03:22:54.069953+0800 0x6280     Default     0x0                  0      0    kernel: (itlwmx) itlwm: : Sorry - debug buffer is only 1024K while you requested 4096K
2020-06-13 03:22:54.069998+0800 0x6280     Default     0x0                  0      0    kernel: (itlwmx) itlwm: iwx_enable_fwload_interrupt
2020-06-13 03:22:55.075419+0800 0x6280     Default     0x0                  0      0    kernel: (itlwmx) itlwm: : could not load firmware
2020-06-13 03:22:55.075889+0800 0x6280     Default     0x0                  0      0    kernel: (itlwmx) : failed to load init firmware
2020-06-13 03:22:55.075892+0800 0x6280     Default     0x0                  0      0    kernel: (itlwmx) itlwm: iwx_stop_device
2020-06-13 03:22:55.175828+0800 0x6280     Default     0x0                  0      0    kernel: (itlwmx) itlwm: iwx_prepare_card_hw
2020-06-13 03:22:55.175832+0800 0x6280     Default     0x0                  0      0    kernel: (itlwmx) itlwm: iwx_set_hw_ready
2020-06-13 03:22:55.175893+0800 0x6280     Default     0x0                  0      0    kernel: (itlwmx) itlwm: attach to interface fail
`

zxystd, 2020-06-13:
有更完整的日志么?因为我这边没有ax201可以测试.

lsvih, 2020-06-13:
同02f0

[itlwmx.log](https://github.com/zxystd/itlwm/files/4773880/itlwmx.log)

htmambo, 2020-06-13:
这个是使用今天论坛上发布的包生成的log
[itlwmx.log](https://github.com/zxystd/itlwm/files/4774622/itlwmx.log)
could not load firmware
failed to load init firmware

zxystd, 2020-07-15:
我最近也开始着手解决这个问题，我有点怀疑cnvio2接口的网卡是不是不能单独工作。对了，这个issue我先关了，已经有同样的issue了，请到这里讨论。#127

closed 2020-07-15 by zxystd