OpenIntelWireless/itlwm #132: "could not load firmware chunk 0 (error 1)"

wdigger, 2020-06-13:
1. Kext is loaded but firmware loading failed with message
`could not load firmware chunk 0 (error 1)`
[log.txt](https://github.com/zxystd/itlwm/files/4774599/log.txt)

2. Unload failed with messages
`(kernel)     Kext com.zxystd.itlwm class itlwm has 1 instance.
(kernel)     Kext com.zxystd.itlwm class CTimeout has 2 instances.
Kernel error handling kext request - (libkern/kext) kext is in use or retained (cannot unload).
Failed to unload com.zxystd.itlwm - (libkern/kext) kext is in use or retained (cannot unload).`

wdigger, 2020-06-13:
And could you add loading, unloading, logging shell commands examples to description.
I use this for obtain log:
`log show --last boot --predicate 'process == "kernel"' | grep -i itlwm > ~/Desktop/log.txt`

zxystd, 2020-06-14:
What's your product id? you can get it from Hackintool.

wdigger, 2020-06-14:
Description: Wireless-AC 9560
Vendor: 0x8086
Device: 0xA370
Sub Device 0x0034
IOReg Path: IOService:/AppleACPIPlatformExpert/PCI0@0/AppleACPIPCI/CNVW@14,3
Device Path: PciRoot(0x0)/Pci(0x14,0x3)

zxystd, 2020-07-15:
warm reboot issue on ac9560, see here #131

closed 2020-07-15 by zxystd