OpenIntelWireless/itlwm #134: as200 working as ethernet

lacostey86, 2020-06-13:
I installed the itlwmx.kext and fill the ssid and pwd ,its working fine , but the issue is its appear as ethernet not a wifi adapter which is now I have 2 ethernet !! , is there any way to change it to wifi adapter ?

zxystd, 2020-06-14:
In this way, the native Wi Fi interface will not be considered in a short time, which is closed source and violates the apple protocol.

Maidenxxx, 2020-06-14:
First I like to thank you @zxystd for all the good work! with this my Intel AX200 work as Ethernet adapter! Second it only work at 2.4 GHz and not in full speed! on 5 Ghz they work very bad...

zxystd, 2020-06-14:
> First I like to thank you @zxystd for all the good work! with this my Intel AX200 work as Ethernet adapter! Second it only work at 2.4 GHz and not in full speed! on 5 Ghz work very bad...

adjust 5 Ghz channel width to 20mhz or auto, not above 20mhz.

closed 2020-07-08 by zxystd

marcelloinfoweb, 2020-07-08:
I downloaded today and compiled, and the same thing happens.