OpenIntelWireless/itlwm #135: Intergrate HeliPort

zxystd, 2020-06-14:
Intergrate HeliPort

Stevemoretz, 2020-06-23:
Hi thanks for all the efforts,After this is done will we get complete ui interaction?Or it's just a some basic integration only?

williambj1, 2020-06-23:
What's your definition/expectation of "complete" and "basic"?

Stevemoretz, 2020-06-23:
Well, I mean can we get a working ui for connecting and disconnecting turning on and off wifi?Or it's just some preparation only and we get these things later?because in your repo for heliport you have written that it's on its first development stages yet.So I was wondering what will we get when this task by zxystd will be done.

williambj1, 2020-06-23:
I'd encourage you to build HeliPort and test it yourself, so far "a working ui for connecting and disconnecting turning on and off wifi" is already implemented.

closed 2020-07-08 by zxystd