OpenIntelWireless/itlwm #136: 崩溃，卡死，错误

lifsux, 2020-06-14:
panic(cpu 0 caller 0xffffff800244aa3a): Kernel trap at 0xffffff800240b580, type 14=page fault, registers:
CR0: 0x000000008001003b, CR2: 0xffffff8117554000, CR3: 0x00000000194aa000, CR4: 0x00000000003626e0
RAX: 0xffffff8117553bd0, RBX: 0x0000000000000001, RCX: 0x0000000000000428, RDX: 0xfffffffffffffff8
RSP: 0xffffff8117553a58, RBP: 0xffffff8117553a60, RSI: 0x0000000000000006, RDI: 0x0000000000000007
R8:  0xffffff8117553bd7, R9:  0x0000000000000007, R10: 0xffffff8117553bd7, R11: 0xfffffffffffffffd
R12: 0x0000000000000000, R13: 0x0000000000000000, R14: 0xffffff8027ee9e40, R15: 0xffffff80e64b2000
RFL: 0x0000000000010282, RIP: 0xffffff800240b580, CS:  0x0000000000000008, SS:  0x0000000000000000
Fault CR2: 0xffffff8117554000, Error code: 0x0000000000000002, Fault CPU: 0x0, PL: 0, VF: 10

Backtrace (CPU 0), Frame : Return Address
0xffffff81175534b0 : 0xffffff800231f5cd 
0xffffff8117553500 : 0xffffff8002458b05 
0xffffff8117553540 : 0xffffff800244a68e 
0xffffff8117553590 : 0xffffff80022c5a40 
0xffffff81175535b0 : 0xffffff800231ec97 
0xffffff81175536b0 : 0xffffff800231f087 
0xffffff8117553700 : 0xffffff8002ac27cc 
0xffffff8117553770 : 0xffffff800244aa3a 
0xffffff81175538f0 : 0xffffff800244a738 
0xffffff8117553940 : 0xffffff80022c5a40 
0xffffff8117553960 : 0xffffff800240b580 
0xffffff8117553a60 : 0xffffff7f857b36b5 
0xffffff8117553ab0 : 0xffffff7f857b2e73 
0xffffff8117553de0 : 0x0 
      Kernel Extensions in backtrace:
         com.zxystd.itlwm(1.0)[A2F6B5BB-09E7-3616-832E-F96758EF2E86]@0xffffff7f85781000->0xffffff7f866c7fff
            dependency: com.apple.iokit.IONetworkingFamily(3.4)[8119788F-2497-3C32-8325-C8B8D843BB31]@0xffffff7f82c65000
            dependency: com.apple.iokit.IOPCIFamily(2.9)[BE052F4D-9B80-3FCD-B36D-BACB7DEE0DF2]@0xffffff7f82d31000

BSD process name corresponding to current thread: kernel_task
Boot args: darkwake=0 gfxrst=1 igfxonln=1 igfxonlnfbs=0x010203

Mac OS version:
Not yet set

Kernel version:
Darwin Kernel Version 19.5.0: Tue May 26 20:41:44 PDT 2020; root:xnu-6153.121.2~2/RELEASE_X86_64
Kernel UUID: 54F1A78D-6F41-32BD-BFED-4381F9F6E2EF
Kernel slide:     0x0000000002000000
Kernel text base: 0xffffff8002200000
__HIB  text base: 0xffffff8002100000
System model name: MacBookPro14,1 (Mac-B4831CEBD52A0C4C)
System shutdown begun: NO
Panic diags file available: NO (0xe00002d8)

System uptime in nanoseconds: 1207338031

lifsux, 2020-06-14:
[itlwm_log.txt](https://github.com/zxystd/itlwm/files/4775926/itlwm_log.txt)

lifsux, 2020-06-14:
[itlwm_log.txt](https://github.com/zxystd/itlwm/files/4775928/itlwm_log.txt)
过去1小时的，网络上传文件用的多

zxystd, 2020-06-14:
哪个版本?

zxystd, 2020-06-14:
这是你强行关机重启以后的错误日志吗?

lifsux, 2020-06-14:
> 哪个版本?

0607

lifsux, 2020-06-14:
> 这是你强行关机重启以后的错误日志吗?

卡死，上不了网，我强制重启了。网络不显示断开，但是打不开网页，假死。

shomerchang, 2020-07-16:
> > 这是你强行关机重启以后的错误日志吗?
> 
> 卡死，上不了网，我强制重启了。网络不显示断开，但是打不开网页，假死。

我是下载速度特别高的时候 会出现这种情况，更新系统时大概10+M/s，  不一会整个系统就卡住，只能强制关机

zxystd, 2020-07-17:
emmm，什么卡？8265？

lifsux, 2020-07-17:
> emmm，什么卡？8265？

是的

shomerchang, 2020-07-20:
> emmm，什么卡？8265？

我有两个本，一个8260, 一个9560， 8260下载速度过高时就会断网，9560速度过高 就会卡死

zxystd, 2020-09-09:
fix in 7d27457, you can try the latest version.

closed 2020-09-09 by zxystd