OpenIntelWireless/itlwm #139: AC9560是不是还是不能驱动呢？

supdo, 2020-06-16:
AC9560是不是还是不能驱动呢？

weidongbcn, 2020-06-18:
可以啊。 
[image]
[image]
[image]

chenyuyou, 2020-06-19:
如果只安装mac，那么AC9560可以用。但是如果重启进入win10后，在回到mac，网卡就不能用了。

lilmik, 2020-06-20:
还是老问题，从win10，linux等系统直接重启进入mac就会出现网卡失效问题。
老老实实关机后重新启动，就没问题。

ErrorErrorError, 2020-06-20:
Also have the same issue, looks like the only was I was able to access the Wi-Fi card is by shutting down my laptop completely and then starting macOS.

allanwhledu, 2020-06-24:
请问一下，在macOS 10.14上AC9560可以用了吗？我试了下还是不行呢，在网络中一直没有以太网的选项可以添加

启动日志也只有：
2020-06-24 05:14:56.361506+0800 0x484      Default     0x0                  0      0    kernel: (kernel) itlwm: free
2020-06-24 05:14:56.362479+0800 0x482      Default     0x0                  0      0    kernel: (kernel) itlwm: free
2020-06-24 05:14:56.366004+0800 0x47e      Default     0x0                  0      0    kernel: (kernel) itlwm: free

llkhs, 2020-06-24:
网速方面怎么样呀！！！

llkhs, 2020-06-28:
用HeliPort是吗

foreveryyl, 2020-07-08:
从windows热重启就可以了。

weidongbcn, 2020-07-08:
我从windows热重启还是不可以。

foreveryyl, 2020-07-14:
> 我从windows热重启还是不可以。

是需要冷开机，写错了

XHL669, 2020-07-15:
除了强制关机如何冷启动呢？我这直接从Win关机，然后按电源键进入macOS还是不行啊。。。
已经把安全启动和快速启动关掉了

williambj1, 2020-07-15:
试试关机后插拔电源

XHL669, 2020-07-15:
> 试试关机后插拔电源

好的，多谢，我试一试。昨天晚上进入windows关机确实插了一宿的电。
还记得我讲的故事吗，终于找到原因了。。。🤣

williambj1, 2020-07-15:
😂

williambj1, 2020-07-17:
fixed in 1fc63d576975d56578ce2048f00f83e7ca550d97

closed 2020-07-17 by williambj1