OpenIntelWireless/itlwm #140: called when not gated

agassecond, 2020-06-16:
For over 2 months i am coming here and there and still cannot get by this error.
IT happens on both itlwm and itlwmx. Card iss ac 9560  8086 02F0

2020-06-16 21:29:20.745419+0300 0x2ea61    Default     0x0                  0      0    kernel: (kernel) Lilu   patcher: @ (DBG) invoked at kext loading/unloading
2020-06-16 21:29:20.745429+0300 0x2ea61    Default     0x0                  0      0    kernel: (kernel) Lilu   patcher: @ (DBG) last kext is 0xFFFFFF7FA206C000 and its name is com.zxystd.itlwmx
2020-06-16 21:29:20.746592+0300 0x2ea80    Default     0x0                  0      0    kernel: PXSX: match category IODefaultMatchCategory exists
2020-06-16 21:29:20.746628+0300 0x2ea74    Default     0x0                  0      0    kernel: XHC: match category IODefaultMatchCategory exists
2020-06-16 21:29:20.747167+0300 0x2ea7b    Default     0x0                  0      0    kernel: RP13: match category IODefaultMatchCategory exists
2020-06-16 21:29:20.748493+0300 0x2ea7a    Default     0x0                  0      0    kernel: RP05: match category IODefaultMatchCategory exists
2020-06-16 21:29:20.748778+0300 0x2ea77    Default     0x0                  0      0    kernel: I2C0: match category IODefaultMatchCategory exists
2020-06-16 21:29:20.748992+0300 0x2ea7e    Default     0x0                  0      0    kernel: SBUS: match category IODefaultMatchCategory exists
2020-06-16 21:29:20.752133+0300 0x2ea61    Default     0x0                  2806   0    kextutil: (Security) [com.apple.securityd:security_exception] MacOS error: -67030
2020-06-16 21:29:20.752768+0300 0x2ea61    Default     0x0                  2806   0    kextutil: (Security) [com.apple.securityd:security_exception] MacOS error: -67030
2020-06-16 21:29:20.763168+0300 0x2ea61    Default     0x0                  2806   0    kextutil: (Security) [com.apple.securityd:security_exception] MacOS error: -67030
2020-06-16 21:29:20.764373+0300 0x2ea61    Default     0x0                  2806   0    kextutil: (Security) [com.apple.securityd:security_exception] MacOS error: -67030
2020-06-16 21:29:20.764990+0300 0x2ea61    Default     0x0                  2806   0    kextutil: (Security) [com.apple.securityd:security_exception] MacOS error: -67030
2020-06-16 21:29:20.766200+0300 0x2ea61    Default     0x0                  2806   0    kextutil: (Security) [com.apple.securityd:security_exception] MacOS error: -67030
2020-06-16 21:29:20.766589+0300 0x2ea61    Default     0x0                  2806   0    kextutil: (Security) [com.apple.securityd:security_exception] MacOS error: -67030
2020-06-16 21:29:20.864285+0300 0x2ea76    Default     0x0                  0      0    kernel: (itlwmx) ieee80211_ifattach
2020-06-16 21:29:20.864294+0300 0x2ea76    Default     0x0                  0      0    kernel: (itlwmx) itlwm: ieee80211_channel_init
2020-06-16 21:29:20.864302+0300 0x2ea76    Default     0x0                  0      0    kernel: (itlwmx) itlwm: ieee80211_set_link_state nstate=2, old_state=-1
2020-06-16 21:29:20.864304+0300 0x2ea76    Default     0x0                  0      0    kernel: (itlwmx) itlwm: ieee80211_set_link_state LINK_STATE_IS_DOWN
2020-06-16 21:29:20.864307+0300 0x2ea76    Default     0x0                  0      0    kernel: (itlwmx) itlwm: ieee80211_media_init
2020-06-16 21:29:20.864311+0300 0x2ea76    Default     0x0                  0      0    kernel: (itlwmx) itlwm: iwx_prepare_card_hw
2020-06-16 21:29:20.864313+0300 0x2ea76    Default     0x0                  0      0    kernel: (itlwmx) itlwm: iwx_set_hw_ready
2020-06-16 21:29:20.864327+0300 0x2ea83    Default     0x0                  0      0    kernel: (itlwmx) itlwm: taskq taskq_create_thread lock
2020-06-16 21:29:20.864328+0300 0x2ea76    Default     0x0                  0      0    kernel: (itlwmx) itlwm: iwx_start_hw
2020-06-16 21:29:20.864330+0300 0x2ea83    Default     0x0                  0      0    kernel: (itlwmx) itlwm: taskq taskq_create_thread unlock
2020-06-16 21:29:20.864330+0300 0x2ea76    Default     0x0                  0      0    kernel: (itlwmx) itlwm: iwx_prepare_card_hw
2020-06-16 21:29:20.864332+0300 0x2ea76    Default     0x0                  0      0    kernel: (itlwmx) itlwm: iwx_set_hw_ready
2020-06-16 21:29:20.864336+0300 0x2ea83    Default     0x0                  0      0    kernel: (itlwmx) itlwm: taskq taskq_create_thread lock
2020-06-16 21:29:20.864338+0300 0x2ea83    Default     0x0                  0      0    kernel: (itlwmx) itlwm: taskq taskq_create_thread unlock
2020-06-16 21:29:20.864343+0300 0x2ea84    Default     0x0                  0      0    kernel: (itlwmx) itlwm: taskq taskq_create_thread lock
2020-06-16 21:29:20.864345+0300 0x2ea84    Default     0x0                  0      0    kernel: (itlwmx) itlwm: taskq taskq_create_thread unlock
2020-06-16 21:29:20.864345+0300 0x2ea85    Default     0x0                  0      0    kernel: (itlwmx) itlwm: taskq taskq_thread schedule task
2020-06-16 21:29:20.864349+0300 0x2ea84    Default     0x0                  0      0    kernel: (itlwmx) itlwm: taskq taskq_create_thread lock
2020-06-16 21:29:20.864352+0300 0x2ea84    Default     0x0                  0      0    kernel: (itlwmx) itlwm: taskq taskq_create_thread unlock
2020-06-16 21:29:20.864353+0300 0x2ea86    Default     0x0                  0      0    kernel: (itlwmx) itlwm: taskq taskq_thread schedule task
2020-06-16 21:29:20.870351+0300 0x2ea76    Default     0x0                  0      0    kernel: (itlwmx) itlwm: : L1 Disabled - LTR Enabled
2020-06-16 21:29:20.870593+0300 0x2ea76    Default     0x0                  0      0    kernel: (itlwmx) itlwm: iwx_run_init_mvm_ucode
2020-06-16 21:29:20.870597+0300 0x2ea76    Default     0x0                  0      0    kernel: (itlwmx) itlwm: iwx_load_ucode_wait_alive
2020-06-16 21:29:20.871571+0300 0x2ea76    Default     0x0                  0      0    kernel: (itlwmx) itlwm: load firmware done
2020-06-16 21:29:20.871593+0300 0x2ea76    Default     0x0                  0      0    kernel: (itlwmx) itlwm: parsing firmware iwlwifi-QuZ-a0-hr-b0-48.ucode
2020-06-16 21:29:20.871616+0300 0x2ea76    Default     0x0                  0      0    kernel: (itlwmx) itlwm: : ucode type 0 section 0
2020-06-16 21:29:20.871623+0300 0x2ea76    Default     0x0                  0      0    kernel: (itlwmx) itlwm: : ucode type 0 section 1
2020-06-16 21:29:20.871630+0300 0x2ea76    Default     0x0                  0      0    kernel: (itlwmx) itlwm: : ucode type 0 section 2
2020-06-16 21:29:20.871638+0300 0x2ea76    Default     0x0                  0      0    kernel: (itlwmx) itlwm: : ucode type 0 section 3
2020-06-16 21:29:20.871645+0300 0x2ea76    Default     0x0                  0      0    kernel: (itlwmx) itlwm: : ucode type 0 section 4
2020-06-16 21:29:20.871652+0300 0x2ea76    Default     0x0                  0      0    kernel: (itlwmx) itlwm: : ucode type 0 section 5
2020-06-16 21:29:20.871659+0300 0x2ea76    Default     0x0                  0      0    kernel: (itlwmx) itlwm: : ucode type 0 section 6
2020-06-16 21:29:20.871666+0300 0x2ea76    Default     0x0                  0      0    kernel: (itlwmx) itlwm: : ucode type 0 section 7
2020-06-16 21:29:20.871673+0300 0x2ea76    Default     0x0                  0      0    kernel: (itlwmx) itlwm: : ucode type 0 section 8
2020-06-16 21:29:20.871681+0300 0x2ea76    Default     0x0                  0      0    kernel: (itlwmx) itlwm: : ucode type 0 section 9
2020-06-16 21:29:20.871688+0300 0x2ea76    Default     0x0                  0      0    kernel: (itlwmx) itlwm: : ucode type 0 section 10
2020-06-16 21:29:20.871695+0300 0x2ea76    Default     0x0                  0      0    kernel: (itlwmx) itlwm: : ucode type 0 section 11
2020-06-16 21:29:20.871702+0300 0x2ea76    Default     0x0                  0      0    kernel: (itlwmx) itlwm: : ucode type 0 section 12
2020-06-16 21:29:20.871713+0300 0x2ea76    Default     0x0                  0      0    kernel: (itlwmx) itlwm: : ucode type 0 section 13
2020-06-16 21:29:20.871720+0300 0x2ea76    Default     0x0                  0      0    kernel: (itlwmx) itlwm: : ucode type 0 section 14
2020-06-16 21:29:20.871729+0300 0x2ea76    Default     0x0                  0      0    kernel: (itlwmx) itlwm: : ucode type 0 section 15
2020-06-16 21:29:20.871736+0300 0x2ea76    Default     0x0                  0      0    kernel: (itlwmx) itlwm: : ucode type 0 section 16
[... cut ...]
2020-06-16 21:29:22.305833+0300 0x2ea76    Default     0x0                  0      0    kernel:          com.zxystd.itlwmx(1.0)[C49B9248-B9EB-3E4B-96C3-C8A50DEB25D5]@0xffffff7f8366c000->0xffffff7f83a45fff
2020-06-16 21:29:22.305847+0300 0x2ea76    Default     0x0                  0      0    kernel:             dependency: com.apple.iokit.IONetworkingFamily(3.4)[8119788F-2497-3C32-8325-C8B8D843BB31]@0xffffff7f80d6f000
2020-06-16 21:29:22.305857+0300 0x2ea76    Default     0x0                  0      0    kernel:             dependency: com.apple.iokit.IOPCIFamily(2.9)[BE052F4D-9B80-3FCD-B36D-BACB7DEE0DF2]@0xffffff7f810b3000

agassecond, 2020-06-25:
Hope it may help,
found in linux ubuntu 20.04 dmesg
[    2.791915] Intel(R) Wireless WiFi driver for Linux
[    2.791915] Copyright(c) 2003- 2015 Intel Corporation
[    2.797421] iwlwifi 0000:00:14.3: Direct firmware load for iwlwifi-QuZ-a0-jf-b0-50.ucode failed with error -2
[    2.797432] iwlwifi 0000:00:14.3: Direct firmware load for iwlwifi-QuZ-a0-jf-b0-49.ucode failed with error -2
[    2.800993] iwlwifi 0000:00:14.3: TLV_FW_FSEQ_VERSION: FSEQ Version: XXX.XXX.XXX.XXX

iwlwifi 0000:00:14.3: loaded firmware version 48.4fa0041f.0 op_mode iwlmvm

once again, it is ac9560 8086 02F0

zxystd, 2020-06-25:
what happened?

agassecond, 2020-06-25:
It  worked on Linux, but still cannot make it work on mac (

ErrorErrorError, 2020-06-25:
Do you have both itlwm.kext and tlwmx.kext loaded? You should be using only itlwm.kext since you have a intel 9560 card. After you use the correct kext, do a full shutdown and load into Mac.

agassecond, 2020-06-28:
zxystd said in other topic:
According to the linux code, 0x02f0 is a gen2 device, which is same pic id as ax201, so, just wait the itlwmx work done.

so, seems waiting more (

zxystd, 2020-07-08:
duplicate issue，close it.

closed 2020-07-08 by zxystd