OpenIntelWireless/itlwm #142: No ipv6 support

Aldo97, 2020-06-17:
Hello,
I tried connecting to a ipv6 address and I wasn't able to do it, in the network settings I see that I have no ipv6 address assigned from the router and even manually assigning it (using the one I have on Windows on the same machine) ipv6 connection is not avaiable.
I have itlwm and itl80211 compiled with the last commits and installed in the clover kext folder, I have no problems with ipv4 and my intel card is the 3165.
Thanks for your work.

williambj1, 2020-06-18:
IPV6 works perfectly as expected on my 9560ac with my router TUF-AX3000

You may probably consider checking your macOS internet configuration/preferences and troubleshoot this issue with HoRNDIS with an Android phone or use a USB wifi card.

Aldo97, 2020-06-18:
I have no issue with HoRNIS and using the host network of the same router I have ipv6 connection. 
I'll try reinstalling osx. 
When I first installed itlwm I had no ipv6 connection available on the router.

Aldo97, 2020-06-19:
I

closed 2020-06-19 by Aldo97