OpenIntelWireless/itlwm #144: 与更新过程出现冲突

MEMZSONBILI, 2020-06-20:
本人在macOS 10.15.5 19F96版本下正常使用，但在更新至19F101时无法进入更新过程，（跑条一点就重启），出现内核恐慌
日志如下:
https://paste.ubuntu.com/p/Z5Y7By3mYv/

williambj1, 2020-06-20:
```
Kernel Extensions in backtrace:
         com.zhong.AppleIntelWiFi(1.0)[34040681-73CD-3F7B-8CBD-E5B15A41224F]@0xffffff7f920f6000->0xffffff7f9219ffff
```

呃呵？

MEMZSONBILI, 2020-06-20:
> ```
> Kernel Extensions in backtrace:
>          com.zhong.AppleIntelWiFi(1.0)[34040681-73CD-3F7B-8CBD-E5B15A41224F]@0xffffff7f920f6000->0xffffff7f9219ffff
> ```
> 
> 呃呵？

我是从 https://b23.tv/VhZvE7 来的 视频介绍有写
如果有一定开发能力的可以访问GitHub对原始驱动二次开发使用
GitHub上面的链接：https://github.com/zxystd/itlwm
该开发者更新速度快于远景论坛
所以我是不是来错地方了？

Goshin, 2020-06-20:
@MEMZSONBILI 从日志看你使用的不是这个仓库的驱动。这个仓库的驱动对应的说明帖子在 http://bbs.pcbeta.com/viewthread-1848662-1-1.html ，其他人重新打包的版本和加载脚本不由该作者维护。

closed 2020-06-22 by zxystd