OpenIntelWireless/itlwm #146: AX200的使用教程

M69W, 2020-06-21:
### [远景链接](http://bbs.pcbeta.com/forum.php?mod=viewthread&tid=1848662)

## 驱动 和 客户端 
### kexts
[itlwm源码](https://github.com/zxystd/itlwm)

上面的远景链接下 `itlwmx_ax200_0618.kext.zip` (`是itlwmx 带x的才适用AX200`)
### HeliPort
[HeliPort源码](https://github.com/zxystd/HeliPort)

#### 下面的编译好的驱动和客户端(`来源黑苹果Intel WiFi交流群 群号:1036194654`)
[HeliPort.zip](https://github.com/zxystd/itlwm/files/4808904/HeliPort.zip)
[itlwmx.kext.zip](https://github.com/zxystd/itlwm/files/4808905/itlwmx.kext.zip)

## 使用
如果不使用HeliPort客户端，记得先改`itlwmx.kext`里面的Info.plist预先写好的SSID和password，
【itlwmx.kext->右键->显示包内容】:`Contents/Info.plist`
[image]
### PS: 执行命令再改info.plist会提示权限问题

根据作者@zxystd在远景上面的链接帖子提供了使用方法，这里暂时使用命令输入启动驱动，如下
```
如果在使用过程中需要添加Wi-Fi但是又不想重启,可以用以下命令卸载,然后编译好之后再次加载
卸载: sudo kextunload -b com.zxystd.itlwm
加载:
sudo chown -R root:wheel itlwm.kext
sudo kextutil -v 6 itlwm.kext
```

这里有必要说下驱动的驱动和路径，
AX200用`itlwmx.kext`
例如驱动放桌面，
在终端输入`sudo chown -R root:wheel `再把`itlwmx.kext`在拖入终端
最后显示就是`sudo chown -R root:wheel /Users/abc/Desktop/itlwmx.kext` 
而不是`sudo chown -R root:wheel itlwm.kext`

使用一般只需要只支持两行命令
```
sudo chown -R root:wheel itlwmx.kext
sudo kextutil -v 6 itlwmx.kext
```

执行命令后，打开 【 系统偏好设置->网络->左下角->【+ 】-> 选【以太网】->应用
[image]

然后打开客户端 HeliPort，第一次没有显示Wi-Fi列表，可尝试多次点击Wi-Fi图标
目前客户端HeliPort 支持Wi-Fi 支持输入Wi-Fi密码，其他功能没测试！

如果没有【以太网】选项，以[小新pro13](https://github.com/daliansky/XiaoXinPro-13-hackintosh) OC为例，
注释或删除两项：
```
DeviceProperties\Add\PciRoot(0x0)/Pci(0x1C,0x0)
DeviceProperties\Add\PciRoot(0x0)/PciRoot(0x0)/Pci(0x1C,0x0)/Pci(0x0,0x0)
```

zncs520, 2020-06-21:
AX201 现在还不行吗?

williambj1, 2020-06-21:
HeliPort 按住 ALT 有退出选项
终端执行 ```sudo killall HeliPort``` 也行

M69W, 2020-06-21:
> HeliPort 按住 ALT 有退出选项
> 终端执行 `sudo killall HeliPort` 也行

多谢提醒

lacostey86, 2020-06-22:
thanks its working fine with ax200 10.15.5 os , but there is one tiny issue if you are already connected to a ssid there is not way to find which ssid i'm using

williambj1, 2020-06-22:
> thanks its working fine with ax200 10.15.5 os , but there is one tiny issue if you are already connected to a ssid there is no way to find which ssid i'm using

This is not an issue, but a feature that hasn't been implemented

arvinws, 2020-06-22:
AX201可以使用么

evertjr, 2020-06-22:
Can you upload the itlwm.kext too?

lawman21c, 2020-06-22:
有个问题，在mac使用了ax200之后，windows10里用不了wifi了。

lacostey86, 2020-06-22:
> > thanks its working fine with ax200 10.15.5 os , but there is one tiny issue if you are already connected to a ssid there is no way to find which ssid i'm using
> 
> This is not an issue, but a feature that hasn't been implemented

ok thanks for the clarification , good job , hope this is the first step for handoff and airdrop

naveenkrdy, 2020-06-23:
great work ! 
AX200 works fine on macos 10.15.5 
@lawman21c  works fine for me on windows even with this kext.

allanwhledu, 2020-06-24:
请问一下，在macOS 10.14上AC9560可以用了吗？我试了下还是不行呢，在网络中一直没有以太网的选项可以添加，是否不支持10.14系统了？

启动日志也只有：
2020-06-24 05:14:56.361506+0800 0x484 Default 0x0 0 0 kernel: (kernel) itlwm: free
2020-06-24 05:14:56.362479+0800 0x482 Default 0x0 0 0 kernel: (kernel) itlwm: free
2020-06-24 05:14:56.366004+0800 0x47e Default 0x0 0 0 kernel: (kernel) itlwm: free

zxystd, 2020-06-24:
支持10.14，你先尝试彻底断电，然后再开启电脑

allanwhledu, 2020-06-24:
> 支持10.14，你先尝试彻底断电，然后再开启电脑

嗯嗯，可以了。其实是因为AC9560有2代的有1代的是吧，我以为都是2代的呢……从itlwmx换回itlwm就没问题了，感谢大佬！！

andiest13, 2020-06-26:
谢谢，可以使用ax200上网了，但是那个网速好像限制了30Mbps左右（转win是可以跑满wifi6的）。

M69W, 2020-06-26:
> 谢谢，可以使用ax200上网了，但是那个网速好像限制了30Mbps左右（转win是可以跑满wifi6的）。

先解决能不能用的问题，
再解决好不好的问题。

继续关注吧

jxp98, 2020-07-02:
貌似还是不行额...是不是只能支持部分ID的AX200

jxp98, 2020-07-03:
同AX200，失败了...一直不显示wifi 直接在kext里该wifi密码也找不到wifi 连不上。大佬，还有得救吗

lawman21c, 2020-07-05:
> 有个问题，在mac使用了ax200之后，windows10里用不了wifi了。

这个问题应该是windows10的，而不是这个驱动的问题。尝试重装系统看看。

leoli-dev, 2020-07-14:
大佬们，AX201 有希望吗？我板载那块是 AX201NGW，这个方法不行吧？

M69W, 2020-07-15:
> 大佬们，AX201 有希望吗？我板载那块是 AX201NGW，这个方法不行吧？

[ax201不能加载 #129](https://github.com/OpenIntelWireless/itlwm/issues/129)

8x13b, 2020-07-17:
Roses are red,
Violets are blue,
The title is in english,
So why is the issue in chinese?

williambj1, 2020-07-17:
I’m working on docs right now, but unfortunately this guide with google translate is currently the best you can get.

scutxd, 2020-07-18:
> 9560

你好，9560还分1代和2代码？如何区分的？

scutxd, 2020-07-18:
> > 支持10.14，你先尝试彻底断电，然后再开启电脑
> 
> 嗯嗯，可以了。其实是因为AC9560有2代的有1代的是吧，我以为都是2代的呢……从itlwmx换回itlwm就没问题了，感谢大佬！！

你好，9560还分1代和2代吗？如何区分呢？

mbirth, 2020-07-20:
Short form:

1. Get Xcode
1. Clone itlwm (from inside Xcode or manually, then open in Xcode)
1. Modify the `itlwm/Info.plist` to match your SSID(s) and password
1. Build/Compile
1. Find the itlwm.kext in `~/Library/Developer/Xcode/DerivedData/itlwm-...../Build/Products/Debug/`
1. Use the Terminal to cd to that dir and try to load the extension to see if it works: `sudo kextload itlwm.kext`
   * probably need to [disable SIP](https://www.howtogeek.com/230424/how-to-disable-system-integrity-protection-on-a-mac-and-why-you-shouldnt/) if you haven't already
1. If it loads fine and your system doesn't crash, you could move it to `/System/Library/Extensions/` to have it autoloaded on boot
   * Catalina+ users will have to remount the system partition for writing first: `sudo mount -uw /`
   * Or move it onto your EFI partition and have OC load it

(If you have Intel cc or Qu/QuZ card, use `itlwmx` instead.)

For HeliPort:

1. Clone repo
1. Have CocoaPods installed (`sudo gem install cocoapods` followed by `pod setup`)
1. cd into cloned dir and install additional cocoa stuff: `pod install`
1. Build/Compile
1. Find HeliPort in `~/Library/Developer/Xcode/DerivedData/HeliPort-...../Build/Products/Debug/`
1. Move it to Applications
1. Set it as a Login Item

For new versions: Do a `git pull` (or from Xcode menu), then clean build dir(!) before atttempting a new build. You may also want to run `pod install` once in a while so updated pods can be pulled in.

Also there's https://github.com/1hbb/OpenIntelWireless-Factory/releases now.

But: **PLEASE DON'T MAKE NEW ISSUES HERE IF YOU DON'T KNOW HOW TO DEBUG PROPERLY!** This is still in development and not yet ready for the general public. If something doesn't work out for you, see if there already is an issue about that (also search the closed issues) and maybe add a comment there, or ask on reddit. Or wait for the next build and see if that fixes your problem.

williambj1, 2020-07-20:
@mbirth Huge thanks for your guide, I would like to slightly modify your comment and correct some errors, is that ok?

mbirth, 2020-07-20:
@williambj1 Sure, go ahead.

williambj1, 2020-07-20:
Here's a more detailed guide for How to build HeliPort

https://openintelwireless.gitbook.io/openintelwireless/heliport/advanced

yoqu, 2020-08-04:
> Roses are red,
> Violets are blue,
> The title is in english,
> So why is the issue in chinese?

Haha...

rexding97, 2020-08-16:
ax200 and use common to run itlwm.kext but fail to load...

dingdongcheng@Dings-iMac-Pro ~ % sudo kextutil -v 6 /Users/dingdongcheng/Desktop/itlwm.kext
Password:
Kext user-space log filter changed from 0xff2 to 0xfff.
Kext kernel-space log filter changed from 0xff2 to 0xfff.
Kext library architecture set to x86_64.
Can't stat /AppleInternal - No such file or directory.
Defaulting to kernel file '/System/Library/Kernels/kernel'
Kext library recording diagnostics for: validation authentication dependencies warnings.
Reading loaded kext info from kernel.
(kernel) User-space log flags changed from 0x0 to 0xfff.
(kernel) Received 'Get Loaded Kext Info' request from user space.
(kernel) Returning loaded kext info.
Loaded kext info:
{
}

Reading loaded kext info from kernel.
(kernel) User-space log flags changed from 0x0 to 0xfff.
(kernel) Received 'Get Loaded Kext Info' request from user space.
(kernel) Returning loaded kext info.
Loaded kext info:
{
}

Kext with invalid signature (-67062) allowed: <OSKext 0x7fc8074129b0 [0x7fff8c66ccc0]> { URL = "file:///Users/dingdongcheng/Desktop/itlwm.kext/", ID = "com.zxystd.itlwm" }
Code Signing Failure: not code signed
/Users/dingdongcheng/Desktop/itlwm.kext appears to be loadable (not including linkage for on-disk libraries).
Reading loaded kext info from kernel.
(kernel) User-space log flags changed from 0x0 to 0xfff.
(kernel) Received 'Get Loaded Kext Info' request from user space.
(kernel) Returning loaded kext info.
Loaded kext info:
{
}

Loading /Users/dingdongcheng/Desktop/itlwm.kext.
Reading load info for 9 kexts.
Reading loaded kext info from kernel.
(kernel) User-space log flags changed from 0x0 to 0xfff.
(kernel) Received 'Get Loaded Kext Info' request from user space.
(kernel) Returning loaded kext info.
Loaded kext info:
{
"com.apple.kpi.bsd" = {
    "OSBundleWiredSize" = 50016
    "CFBundleIdentifier" = "com.apple.kpi.bsd"
    "OSBundleUUID" = <91a52191 d68e41a5 91162683 c2bbd6e8>
    "OSBundleCompatibleVersion" = "8.0.0b1"
    "CFBundleVersion" = "19.6.0"
    "OSBundleStarted" = true
    "OSKernelResource" = true
    "OSBundleLoadTag" = 1
    "OSBundleLoadAddress" = -551890460672
    "OSBundlePath" = 
        "/System/Library/Extensions/System.kext/PlugIns/BSDKernel.kext"
    "OSBundleRetainCount" = 112
    "OSBundlePrelinked" = true
    "OSBundleIsInterface" = true
    "OSBundleLoadSize" = 50016
}
"com.apple.kpi.libkern" = {
    "OSBundleWiredSize" = 65712
    "CFBundleIdentifier" = "com.apple.kpi.libkern"
    "OSBundleUUID" = <2ac6caf5 0e694e40 bcac6f1b 6fe7254d>
    "OSBundleCompatibleVersion" = "8.0.0d0"
    "CFBundleVersion" = "19.6.0"
    "OSBundleStarted" = true
    "OSKernelResource" = true
[... 170 lines cut ...]
(kernel) Recorded kext com.zxystd.itlwm as a candidate for inclusion in prelinked kernel.
(kernel) Loading kext com.zxystd.itlwm.
(kernel) Kext com.apple.kpi.bsd is already loaded.
(kernel) Kext com.apple.kpi.mach is already loaded.
(kernel) Kext com.apple.iokit.IONetworkingFamily is already loaded.
(kernel) Kext com.apple.iokit.IONetworkingFamily sending 2 personalities to the IOCatalogue and starting matching.
(kernel) Kext com.apple.kpi.iokit is already loaded.
(kernel) Kext com.apple.kpi.libkern is already loaded.
(kernel) Kext com.apple.iokit.IOPCIFamily is already loaded.
(kernel) Kext com.apple.iokit.IOPCIFamily sending 3 personalities to the IOCatalogue and starting matching.
(kernel) Allocated link buffer for kext com.zxystd.itlwm at 0xffffff7fa12f2000 (16261120 bytes).
(kernel) Kext com.zxystd.itlwm executable loaded; 3970 pages at 0xffffff7f866f2000 (load tag 149).
(kernel) Kext com.zxystd.itlwm calling module start function.
(kernel) OSMetaClass: Kext com.zxystd.itlwm class pci_intr_handle is a duplicate;kext com.zhong.AppleIntelWiFi already has a class by that name.
(kernel) Kext com.zxystd.itlwm did not start (return code 0xdc00400a).
(kernel) Kext com.zxystd.itlwm start failed (result 0xdc00400a).
(kernel) Kext com.zxystd.itlwm failed to load (0xdc008017).
(kernel) Failed to load kext com.zxystd.itlwm (error 0xdc008017).
(kernel) Kext com.zxystd.itlwm unloading.
(kernel) Kext com.zxystd.itlwm unwiring and unmapping linked executable.
(kernel) Kext com.zxystd.itlwm unloaded.
(kernel) Kext com.zxystd.itlwm removing all personalities naming it from the IOCatalogue.
Kernel error handling kext request - (libkern/kext) kext (kmod) start/stop routine failed.
Failed to load /Users/dingdongcheng/Desktop/itlwm.kext - (libkern/kext) kext (kmod) start/stop routine failed.
Failed to load /Users/dingdongcheng/Desktop/itlwm.kext - (libkern/kext) kext (kmod) start/stop routine failed.

LeoLittleZhun, 2020-08-20:
HeliPort好像有点问题啊。对于不需要密码的wifi也要强制你输入密码。所以没有密码的WiFi就连不上

williambj1, 2020-08-20:
@LeoLittleZhun 请按照 issue 模板打开一个新的 issue

M69W, 2020-08-24:
**08-24-2020更**  ： 
[小新pro13](https://github.com/daliansky/XiaoXinPro-13-hackintosh) 在Big Sur中不禁用 ASPM 注入也能正常使用AX200/AX201
如果为了省电，`AX201`也可以改为以下`CNVW路径`，并设置对应的[ASPM工作模式](https://github.com/daliansky/OC-little/tree/master/01-%E5%85%B3%E4%BA%8EAOAC/01-5-%E8%AE%BE%E7%BD%AEASPM%E5%B7%A5%E4%BD%9C%E6%A8%A1%E5%BC%8F)
```
DeviceProperties\Add\PciRoot(0x0)/Pci(0x14,0x3)
DeviceProperties\Add\PciRoot(0x0)/PciRoot(0x0)/Pci(0x14,0x3)/Pci(0x0,0x0)
```

closed 2020-08-24 by M69W

LingyeHan, 2020-09-27:
这试了wif可以正常使用，就是bluetooth不显示，不知道怎么解决？