OpenIntelWireless/itlwm #150: AC9462 0x42A4 not working

lgariv, 2020-06-24:
I tested this kext when it was in an earlier stage, and added in a pull request (that was later merged) the device id's relevant for AC9462, and AC9461. 
As the title says, mine is 0x8086 0x42A4.
In that build it was working correctly, I was able to connect to my router and use it.
since then, I stopped testing for a while because it was clear that there's a lot more to do, and now that I'm testing it again - it does not work at all. I looked in the issues and found no one else with this card and this device id, so I think it worth to look into it.

In summary:
My card, AC9462 0x8086 0x42A4, WAS working in past commits, and now when loading the kext (either itlwm or itlwmx) nothing happens. my card works with the same firmware and "properties" as the AC9560 - so as long as AC9560 works, my card should work as well.

zxystd, 2020-06-24:
so can you test the previous merge commit again?

lgariv, 2020-06-24:
> so can you test the previous merge commit again?

Yes, just tested again, still works in the old commit.

zxystd, 2020-06-24:
emmm, any logs when loading?

lgariv, 2020-06-24:
> emmm, any logs when loading?

Actually... I think it works? Before, nothing would happen when loading the latest commit. After I loaded the old commit, I tried to unload it and forgot that there was an issue with freezing when unloading back then, so I had to restart my machine. Then I loaded the latest commit, and it works, I am using it to post this comment.

I think that's the real issue: the latest commit doesn't load the device if it is not already in preferences. The old commit added the device correctly to preferences, and after that the latest commit works.

By the way, download speed is around half of what it should be, and upload is perfect. I remember it was the other way around in the past, interesting. Another thing to point out - Wi-Fi network list is pretty short, and includes some networks that I know belong to to people who live at least 2 floors above me, but at the same time it shows me only 1 out of 3 networks that I have in my own home. Unloading the re-loading does change the networks, but it seems rather random and not based on how close they are or how good the reception is.

rafhagunz, 2020-06-24:
> Actually... I think it works? Before, nothing would happen when loading the latest commit. After I loaded the old commit, I tried to unload it and forgot that there was an issue with freezing when [...]

Can you upload the lastest itlwm that u are using for test?
I have the ac9462 and mine is working, and show me a lot of wifi

lgariv, 2020-06-24:
> Can you upload the lastest itlwm that u are using for test?
> I have the ac9462 and mine is working, and show me a lot of wifi

[itlwm.kext.zip](https://github.com/zxystd/itlwm/files/4825113/itlwm.kext.zip)

zxystd, 2020-06-24:
there are two questions.
1.it is a warn reboot issue happens in 9560 series devices, if you are reboot from windows or linux to mac, the other system's wifi driver corrupt the pcie device so it won't work again until you full shutdow computer and restart. now i don't have these device to debug.
2.it is not a issue, the scan results are not random, just because it will not auto scan after connected. will do active scan when tbe connected wifi not well.

zxystd, 2020-07-17:
Now warm reboot issue fixed, you can test the latest code. I am going to close this issue, if any problem exists, please let me know. thanks.

closed 2020-07-17 by zxystd