OpenIntelWireless/itlwm #154: several deauth's happen when having load on upload

h1ghrise, 2020-06-29:
hi,
first, i really appreciate your dedication and work you did here - Simply amazing. Bringing Intel Wifi Support to the mac is a huge step :)

build:
- Matebook 13" (WRT-WX9 M1200)
- Intel i7 8565U / UHD 620
- Intel AC 9560 CNVi Wifi Card
- OpenCore 0.59
- your latest Kext  (Wifi Settings: WPA2 Personal, Alphanumeric SSID, no spaces, entered as WIFI_1 in info.plist)

my Wifi:
```
2020-06-29 16:56:56.128397+0200 0x13f      Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_endscan scan_result ssid=Walhalla, bssid=5c:49:79:d4:49:6c, ni_rsnciphers=8, ni_rsncipher=0, ni_rsngroupmgmtcipher=32, ni_rsngroupcipher=8, ni_rssi=56,  ni_capinfo=4145, ni_intval=100, ni_rsnakms=2, ni_supported_rsnakms=2, ni_rsnprotos=1, ni_supported_rsnprotos=1, ni_rstamp=110899529
```

I used this kext for some days now, and now noticed some odd behavior (maybe somebody can confirm/reproduce?)

- I get deauthe'd when there is more load at the upload
- Mostly everytime I open a Video on Youtube, Packetloss increases drastically. According to kernel logs error packets.
- When deauthe'd sometimes, the connection gets re-established (handshake successful), but sometimes not. In those cases i have to reboot the machine.

Please find Kernel log attached. Let me know if you need to know more.
```➜  ~ log stream | grep itlwm
2020-06-29 16:56:47.484701+0200 0x13f      Default     0x0                  0      0    kernel: (itlwm) itlwm: ieee80211_recv_deauth
2020-06-29 16:56:47.484714+0200 0x13f      Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_newstate
2020-06-29 16:56:47.484719+0200 0x13f      Default     0x0                  0      0    kernel: (itlwm) itlwm: taskq task_add newstate_task add to queue
2020-06-29 16:56:47.484730+0200 0x17a      Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_sf_config
2020-06-29 16:56:47.484739+0200 0x17a      Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_disable_beacon_filter
2020-06-29 16:56:47.486122+0200 0x17a      Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_phy_ctxt_cmd
2020-06-29 16:56:47.486827+0200 0x17a      Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_disassoc
2020-06-29 16:56:47.487445+0200 0x17a      Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_deauth
2020-06-29 16:56:47.489508+0200 0x17a      Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_auth
2020-06-29 16:56:47.489518+0200 0x17a      Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_phy_ctxt_cmd
2020-06-29 16:56:47.493337+0200 0x17a      Default     0x0                  0      0    kernel: (itlwm) itlwm: itlwm: RUN -> AUTH
2020-06-29 16:56:47.493348+0200 0x17a      Default     0x0                  0      0    kernel: (itlwm) itlwm: ieee80211_set_link_state nstate=2, old_state=4
2020-06-29 16:56:47.493352+0200 0x17a      Default     0x0                  0      0    kernel: (itlwm) itlwm: ieee80211_set_link_state LINK_STATE_IS_DOWN
2020-06-29 16:56:47.493408+0200 0x17a      Default     0x0                  0      0    kernel: (itlwm) itlwm: itlwm: sending auth to 5c:49:79:d4:49:6c on channel 1 mode 11n type=176
2020-06-29 16:56:47.497714+0200 0x13f      Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_rx_tx_cmd_single 703 OUTPUT_ERROR status=131
2020-06-29 16:56:51.641842+0200 0x13fdc    Default     0x0                  0      0    kernel: (itlwm) itlwm: itlwm: authentication timed out for 5c:49:79:d4:49:6c
2020-06-29 16:56:51.641859+0200 0x13fdc    Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_newstate
2020-06-29 16:56:51.641863+0200 0x13fdc    Default     0x0                  0      0    kernel: (itlwm) itlwm: taskq task_add newstate_task add to queue
2020-06-29 16:56:51.641877+0200 0x17a      Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_deauth
2020-06-29 16:56:51.643079+0200 0x17a      Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_scan
2020-06-29 16:56:51.643092+0200 0x17a      Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_umac_scan
2020-06-29 16:56:51.643100+0200 0x17a      Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_fill_probe_req_v1
2020-06-29 16:56:51.643102+0200 0x17a      Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_fill_probe_req
2020-06-29 16:56:51.643109+0200 0x17a      Default     0x0                  0      0    kernel: (itlwm) itlwm: large command paylen=1684 len0=1684
2020-06-29 16:56:51.644201+0200 0x17a      Default     0x0                  0      0    kernel: (itlwm) itlwm: ieee80211_reset_scan
2020-06-29 16:56:51.644211+0200 0x17a      Default     0x0                  0      0    kernel: (itlwm) itlwm: itlwm: AUTH -> SCAN
2020-06-29 16:56:51.644214+0200 0x17a      Default     0x0                  0      0    kernel: (itlwm) itlwm: ieee80211_set_link_state nstate=2, old_state=2
2020-06-29 16:56:52.677361+0200 0x13f      Default     0x0                  0      0    kernel: (itlwm) itlwm: ieee80211_setup_node 34:2c:c4:4a:9e:01
2020-06-29 16:56:52.677380+0200 0x13f      Default     0x0                  0      0    kernel: (itlwm) itlwm: ieee80211_setup_node 36:2c:94:4a:9e:01
2020-06-29 16:56:52.677391+0200 0x13f      Default     0x0                  0      0    kernel: (itlwm) itlwm: ieee80211_setup_node 34:2c:c4:2e:77:13
2020-06-29 16:56:52.677458+0200 0x13f      Default     0x0                  0      0    kernel: (itlwm) itlwm: ieee80211_setup_node 92:5c:14:81:5a:d3
2020-06-29 16:56:52.677479+0200 0x13f      Default     0x0                  0      0    kernel: (itlwm) itlwm: ieee80211_setup_node 00:11:32:4d:1c:47
2020-06-29 16:56:52.677487+0200 0x13f      Default     0x0                  0      0    kernel: (itlwm) itlwm: ieee80211_setup_node 06:7c:34:89:ae:03
2020-06-29 16:56:53.920275+0200 0x13f      Default     0x0                  0      0    kernel: (itlwm) itlwm: ieee80211_setup_node 06:7c:34:87:4d:ca
2020-06-29 16:56:56.126882+0200 0x13f      Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_endscan scan_result ssid=SynologySmartConnect, bssid=00:11:32:4d:1c:47, ni_rsnciphers=8, ni_rsncipher=0, ni_rsngroupmgmtcipher=32, ni_rsngroupcipher=8, ni_rssi=20,  ni_capinfo=5137, ni_intval=100, ni_rsnakms=2, ni_supported_rsnakms=2, ni_rsnprotos=1, ni_supported_rsnprotos=1, ni_rstamp=115794218
2020-06-29 16:56:56.126924+0200 0x13f      Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_endscan scan_result ssid=SynologySmartConnect_5G, bssid=00:11:32:4d:1c:48, ni_rsnciphers=8, ni_rsncipher=0, ni_rsngroupmgmtcipher=32, ni_rsngroupcipher=8, ni_rssi=14,  ni_capinfo=4369, ni_intval=100, ni_rsnakms=2, ni_supported_rsnakms=2, ni_rsnprotos=1, ni_supported_rsnprotos=1, ni_rstamp=857880
2020-06-29 16:56:56.126951+0200 0x13f      Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_endscan scan_result ssid=UPC Wi-Free, bssid=06:7c:34:87:4d:ca, ni_rsnciphers=12, ni_rsncipher=0, ni_rsngroupmgmtcipher=32, ni_rsngroupcipher=4, ni_rssi=23,  ni_capinfo=1041, ni_intval=100, ni_rsnakms=1, ni_supported_rsnakms=1, ni_rsnprotos=1, ni_supported_rsnprotos=3, ni_rstamp=116185510
2020-06-29 16:56:56.126982+0200 0x13f      Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_endscan scan_result ssid=UPC Wi-Free, bssid=06:7c:34:89:ae:03, ni_rsnciphers=12, ni_rsncipher=0, ni_rsngroupmgmtcipher=32, ni_rsngroupcipher=4, ni_rssi=15,  ni_capinfo=1041, ni_intval=100, ni_rsnakms=1, ni_supported_rsnakms=1, ni_rsnprotos=1, ni_supported_rsnprotos=3, ni_rstamp=115890447
2020-06-29 16:56:56.127011+0200 0x13f      Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_endscan scan_result ssid=Anderssens, bssid=10:da:43:91:3a:2e, ni_rsnciphers=0, ni_rsncipher=0, ni_rsngroupmgmtcipher=0, ni_rsngroupcipher=0, ni_rssi=18,  ni_capinfo=5137, ni_intval=100, ni_rsnakms=0, ni_supported_rsnakms=0, ni_rsnprotos=0, ni_supported_rsnprotos=0, ni_rstamp=1311246979
2020-06-29 16:56:56.127036+0200 0x13f      Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_endscan scan_result ssid=UPC8170533, bssid=34:2c:c4:2e:76:ef, ni_rsnciphers=8, ni_rsncipher=0, ni_rsngroupmgmtcipher=32, ni_rsngroupcipher=4, ni_rssi=33,  ni_capinfo=5425, ni_intval=100, ni_rsnakms=2, ni_supported_rsnakms=2, ni_rsnprotos=1, ni_suppo [... cut ...]
2020-06-29 16:58:22.265180+0200 0x13f      Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_rx_tx_cmd_single 703 OUTPUT_ERROR status=131
2020-06-29 16:58:22.280747+0200 0x13f      Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_rx_tx_cmd_single 703 OUTPUT_ERROR status=131
2020-06-29 16:58:22.291814+0200 0x13f      Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_rx_tx_cmd_single 703 OUTPUT_ERROR status=131

... like 200 times
```

deauth is also visible in my Router`s log (sorry for the German ;D)
[image]

zxystd, 2020-06-29:
error code 131 means the packet is too large, that is why mtu exists, you can adjust the mtu value in network setting panel, if that can not take effect, you can modify the source code, change 1482 to 1460 and try again. waiting for your response.

[image]

h1ghrise, 2020-06-29:
Hey,
thanks for your prompt reply. The MTU was adjusted to the value in the define (1482).
After reboot and checking (downloading media, youtube - basically everything which produces load on the interface) i did not encounter the error so far. 
Will watch out and reply if there is something to respond.

h1ghrise, 2020-06-30:
Hi again,
after checking and observing for a day, i can draw new conclusions:
- even after i have adjusted the WIFI Settings in the `info.plist` file to 5Ghz Wifi, the kext is not using it. (Still connects to the "old" 2.4Ghz Wifi (even when i have deleted it). Is this stored somewhere?
- Error 131 (MTU issues) are only happening when connected to 2.4Ghz Wifi.

```72672738-2020-06-30 10:26:49.107723+0200 0x13f      Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_rx_tx_cmd_single 703 OUTPUT_ERROR status=131
72672898-2020-06-30 10:26:49.143841+0200 0x13f      Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_rx_tx_cmd_single 703 OUTPUT_ERROR status=131
72673058-2020-06-30 10:26:49.161461+0200 0x13f      Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_rx_tx_cmd_single 703 OUTPUT_ERROR status=131
72673803-2020-06-30 10:26:49.775217+0200 0x4c45     Default     0x0                  0      0    kernel: (itlwm) itlwm: disable
72674068-2020-06-30 10:26:49.775250+0200 0x4c45     Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_stop
72674188-2020-06-30 10:26:49.775253+0200 0x4c45     Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_stop_device
72674315-2020-06-30 10:26:49.829841+0200 0x4c45     Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_conf_msix_hw
72674443-2020-06-30 10:26:49.829864+0200 0x4c45     Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_prepare_card_hw
72674574-2020-06-30 10:26:49.829880+0200 0x4c45     Default     0x0                  0      0    kernel: (itlwm) itlwm: itlwm: RUN -> INIT
72674704-2020-06-30 10:26:49.829884+0200 0x4c45     Default     0x0                  0      0    kernel: (itlwm) itlwm: ieee80211_set_link_state nstate=2, old_state=4
72674862:2020-06-30 10:26:49.829891+0200 0x4c45     Default     0x0                  0      0    kernel: (itlwm) itlwm: ieee80211_set_link_state LINK_STATE_IS_DOWN
```

when checking the SSID:

``` 
241622476-2020-06-30 13:57:07.239323+0200 0x17a      Default     0x0                  0      0    kernel: (itlwm) itlwm: itlwm: sending auth to 5c:49:79:d4:49:6c on channel 1 mode 11n type=176
```
the `:6c` is the 2.4 Ghz one. `6d` would be 5Ghz, which worked later on pretty well.
````
2020-06-30 17:46:21.076674+0200 0x140      Default     0x0                  0      0    kernel: (itlwm) itlwm: itlwm: + 5c:49:79:d4:49:6d
2020-06-30 17:46:21.076856+0200 0x140      Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_endscan scan_result ssid=Walhalla5G, bssid=5c:49:79:d4:49:6d, ni_rsnciphers=8, ni_rsncipher=0, ni_rsngroupmgmtcipher=32, ni_rsngroupcipher=8, ni_rssi=31,  ni_capinfo=5393, ni_intval=100, ni_rsnakms=2, ni_supported_rsnakms=2, ni_rsnprotos=1, ni_supported_rsnprotos=1, ni_rstamp=-719241207
```
connect:
```2020-06-30 18:36:50.523802+0200 0x140      Default     0x0                  0      0    kernel: (itlwm) itlwm: itlwm: received msg 1/2 of the group key handshake from 5c:49:79:d4:49:6d
2020-06-30 18:36:50.523815+0200 0x140      Default     0x0                  0      0    kernel: (itlwm) itlwm: itlwm: sending msg 2/2 of the group key handshake to 5c:49:79:d4:49:6d````

Im checking what difference it makes, when im using the `define ETHERNET_MTU` to a different, fixed value. In which sourcefile is this defined? (did not find it yet;))

Cheers,
Benjamin

zxystd, 2020-06-30:
if you are using the same ssid between 2.4Ghz and 5Ghz, maybe it can not decide which one it should connect and connect one of them randomly. Also, the ETHERNET_MTU value is define in itlwm_interface.hpp

h1ghrise, 2020-06-30:
They have different SSIDs. (the 5g one has 5G as postfix)

2.4Ghz (was entered in info.plist before):
```2020-06-30 17:46:21.076848+0200 0x140      Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_endscan scan_result ssid=Walhalla, bssid=5c:49:79:d4:49:6c, ni_rsnciphers=8, ni_rsncipher=0, ni_rsngroupmgmtcipher=32, ni_rsngroupcipher=8, ni_rssi=37,  ni_capinfo=4145, ni_intval=100, ni_rsnakms=2, ni_supported_rsnakms=2, ni_rsnprotos=1, ni_supported_rsnprotos=1, ni_rstamp=-730499238```

5Ghz (now entered in info.plist):
```2020-06-30 17:46:21.076856+0200 0x140      Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_endscan scan_result ssid=Walhalla5G, bssid=5c:49:79:d4:49:6d, ni_rsnciphers=8, ni_rsncipher=0, ni_rsngroupmgmtcipher=32, ni_rsngroupcipher=8, ni_rssi=31,  ni_capinfo=5393, ni_intval=100, ni_rsnakms=2, ni_supported_rsnakms=2, ni_rsnprotos=1, ni_supported_rsnprotos=1, ni_rstamp=-719241207```

thanks for the hint. I`ll check if this makes any difference.

Edoardo001, 2020-06-30:
> They have different SSIDs. (the 5g one has 5G as postfix)
>
> 2.4Ghz (was entered in info.plist before):
> ```2020-06-30 17:46:21.076848+0200 0x140      Default     0x0                  0      0 [...]

Hey man, scuse for the wrong place, I have same laptop. Have you jack 3,5 or camera working?

h1ghrise, 2020-07-01:
[image]
i captured the situation when connection gets lost. Let me know if you want the pcap :)

zxystd, 2020-07-15:
any update information by using latest kext?

closed 2020-07-17 by zxystd