OpenIntelWireless/itlwm #156: Compiling error on Xcode Catalina with itlwm

mr-vito, 2020-06-30:
I am trying to compile the latest version of itlwm with Xcode 11.5.

All I do is change the config.plist and compile but I get a semantic error, which fails to compile the itlwm.kext.

"Assigning to 'bus_addr_t' (aka 'unsigned long long') from incompatible type 'nullptr_t')"
File in question is io.cpp > itlwm:: iwm_dma_config_free(dma).

Thanks for your help!

ErrorErrorError, 2020-06-30:
Make sure you're using the latest SDK when building.

mr-vito, 2020-06-30:
> Make sure you're using the latest SDK when building.

Thanks for your reply! I am using macOS 10.15 when building, so latest non-beta SDK. Problem persists...

ailuoku6, 2020-07-01:
[image]

Longcoming, 2020-07-01:
I am getting four errors when compiling:
//  itlwm

//  Copyright © 2020 钟先耀. All rights reserved.
#include "FwData.h"

const unsigned char itlwm_master[] = {
};

const long int itlwm_master_size = sizeof(itlwm_master);

const unsigned char *[] = {
};

const long int *_size = sizeof(*);

const struct FwDesc fwList[] = {
{IWL_FW("itlwm-master", itlwm_master, itlwm_master_size)},
{IWL_FW("*", *, *_size)},
};
const int fwNumber = 2;

mr-vito, 2020-07-01:
> I am getting four errors when compiling:
> // itlwm
>
> // Copyright © 2020 钟先耀. All rights reserved.
> #include "FwData.h"
>
> const unsigned char itlwm_master[] = {
> };
>
> const long int itlwm_master_size = [...]

I can also confirm I have now 4 errors as well. Thanks!

Longcoming, 2020-07-01:
ı solved my compiling problem by doing this.
1.open terminal
2.git clone https://github.com/zxystd/itlwm.git
3.locate the itlwm folder(users/your user name)
4.open itlwm
5.first run fwgen then target itlwm

ı hope it helps

mehmetyukcell, 2020-07-01:
> ı solved my compiling problem by doing this.
> 
> 1.open terminal
> 
> 2.git clone https://github.com/zxystd/itlwm.git
> 
> 3.locate the itlwm folder(users/your user name)
> 
> 4.open itlwm
> 
> 5.first run fwgen then target itlwm
> 
> 
> 
> ı hope it helps

Thanks but How do you compiled Heliport?

Longcoming, 2020-07-01:
ı only compiled itlwm.kext which is now not connecting.
I don't know Heliport,ıf you could enlighten me ı would be pleased.

mr-vito, 2020-07-01:
> ı solved my compiling problem by doing this.
> 1.open terminal
> 2.git clone https://github.com/zxystd/itlwm.git
> 3.locate the itlwm folder(users/your user name)
> 4.open itlwm
> 5.first run fwgen then target itlwm
> 
> ı hope it helps

Thanks, this worked. No errors left now.

Although, when I changed the variable to 0 (https://github.com/zxystd/itlwm/issues/156#issuecomment-652113894), cleaned build folder, ran fwgen then itlwm target, it would not compile successfully.

Longcoming, 2020-07-01:
NULL means empty,0 is not empty

Longcoming, 2020-07-01:
can you connect now,because ı can't

vitalpig, 2020-07-01:
> > ı solved my compiling problem by doing this.
> > 1.open terminal
> > 2.git clone https://github.com/zxystd/itlwm.git
> > 3.locate the itlwm folder(users/your user name)
> > 4.open itlwm
> > 5.first run fwgen then target itlwm
> > ı hope it helps
> 
> Thanks but How do you compiled Heliport?

I complied HeliPort by doing this. I don't know whether it is the right way.
1.git clone https://github.com/zxystd/HeliPort.git
2. cd HeliPort
3. pod install
4. open the HeliPort.xcworkspace

Longcoming, 2020-07-01:
how did you install podfile?can you specify?

vitalpig, 2020-07-01:
> how did you install podfile?can you specify?

open the terminal, locate the HeliPort folder , enter the command "pod install"

Longcoming, 2020-07-01:
command not found:pod

Longcoming, 2020-07-01:
after installing cocoapods,ıt worked.thanks.
By the way,heliport works

closed 2020-07-08 by zxystd