OpenIntelWireless/itlwm #157: AC9560 detected but not working

antoniomcr96, 2020-07-01:
Kext loads, from logs it seems that card works (it always scans and finds the network in info.plist).
It doesn’t result connected, though. 
I open heliport and WiFi is listed as currently on, not connected. I let it scan for networks, it correctly finds my WiFi network and connects to it.
Nevertheless I don’t have any Ethernet connection or anything else in Network preferences. So internet doesn’t work.
Thanks for the amazing work, Heliport GUI is very well done

zxystd, 2020-07-01:
collect the log and send it to me.
by the way, what is your card product id?

`log show --last boot | grep itlwm >~/Desktop/itlwm.log`

antoniomcr96, 2020-07-01:
Some news! For some reason after suddenly after a reboot it didn't work: _Wifi: status not available_ in Heliport and Wireless not detected in Hackintool. Cleaning NVRAM didn't help, but `sudo rm /Library/Preferences/SystemConfiguration/NetworkInterfaces.plist` and deleting kext's cache (not sure if useful) fixed the problem this time again.
This is a log of the problem:
[9560log.txt](https://github.com/zxystd/itlwm/files/4859378/9560log.txt)
Hope you can find it useful

mehmetyukcell, 2020-07-01:
> Some news! For some reason suddenly after a reboot it didn't work: _Wifi: status not available_ in Heliport and Wireless not detected in Hackintool. Cleaning NVRAM didn't help, but `sudo rm /Library/Preferences/SystemConfiguration/NetworkInterfaces.plist` and deleting kext's cache (not sure if useful) fixed the problem this time again.
> This is a log of the problem:
> [9560log.txt](https://github.com/zxystd/itlwm/files/4859378/9560log.txt)
> Hope you can find it useful
> 
> 0x9DF0 I think it's the product iD

work same card in the my system

antoniomcr96, 2020-07-02:
I think there are some little problems with usb connections.
I report what I noticed, if useful:
if I boot with an iPhone connected for usb tethering the kext loads and it says connected, even though Ethernet connections doesn't result online and the system actually uses my iPhone. Unplugging the phone doesn't solve the problem and I have to restart.
[9560log2.txt](https://github.com/zxystd/itlwm/files/4861375/9560log2.txt)
It's a bit long. I unplug the phone at 02:29 and I try to connect to the network with Heliport

Matgen84, 2020-07-06:
On Z390 Aorus master (catalina 10.15.5). Same issue: I try to connect to the network with Heliport. It says connected but Ethernet connection isn't. Why?

mehmetyukcell, 2020-07-06:
> On Z390 Aorus master (catalina 10.15.5). Same issue: I try to connect to the network with Heliport. It says connected but Ethernet connection isn't. Why?

kext to clover/kexts/other and app applications and restart and control Hackintool 
[itlwm.kext.zip](https://github.com/OpenIntelWireless/itlwm/files/4878810/itlwm.kext.zip)
[HeliPort.app.zip](https://github.com/OpenIntelWireless/itlwm/files/4878813/HeliPort.app.zip)

Matgen84, 2020-07-06:
> kext to clover/kexts/other and app applications and restart and control Hackintool
> [itlwm.kext.zip](https://github.com/OpenIntelWireless/itlwm/files/4878810/itlwm.kext.zip) [...]

Thanks.

With this kext, this time, KP 
I disable it, I can boot

zxystd, 2020-08-03:
use latest pre-release. https://github.com/OpenIntelWireless/itlwm/releases

Matgen84, 2020-08-03:
> use latest pre-release. https://github.com/OpenIntelWireless/itlwm/releases

Thanks. There is only debug version !

closed 2020-08-15 by williambj1

antoniomcr96, 2020-09-30:
I confirm that latest version seems to have solved the problem. Wifi works well with Heliport, thanks for all your work!