OpenIntelWireless/itlwm #158: how to compile the kext ?

xrokz, 2020-07-02:
When i hit build in Xcode i got the build successes but i cant open the kext folder with `Show in Finder` option i dont know how to get the kexts.

any help ?

schspa, 2020-07-02:
I'm newbie too.  And I found the builded kexts is at the folder 
"/Users/schspa/Library/Developer/Xcode/DerivedData/itlwm-giwpqyjdwchuroetixtxfftvvruv/Build/Products/Release"

igorkulman, 2020-07-03:
run

```bash
xcodebuild -project itlwm.xcodeproj -target itlwm -configuration Release
```
and the kext will be build and located in `./build/Release/`

Matgen84, 2020-07-04:
Hi 
I use terminal command without success:
error: unable to find sdk 'macosx10.12' (in target 'itlwm' from project 'itlwm')

** BUILD FAILED **

Need help. Please

igorkulman, 2020-07-04:
> Hi
> I use terminal command without success:
> error: unable to find sdk 'macosx10.12' (in target 'itlwm' from project 'itlwm')
> 
> ** BUILD FAILED **
> 
> Need help. Please

Change the SDK version in the project from 10.12 to the version of macOS you are using. Or just delete this line https://github.com/zxystd/itlwm/blob/master/itlwm.xcodeproj/project.pbxproj#L1307

Matgen84, 2020-07-04:
> Change the SDK version in the project from 10.12 to the version of macOS you are using. Or just delete this line https://github.com/zxystd/itlwm/blob/master/itlwm.xcodeproj/project.pbxproj#L1307

Thanks a lot

mackonsti, 2020-07-04:
Hi @zxystd can we kindly ask you to change your file **itlwm.xcodeproj/project.pbxproj** and replace inside the 3 instances of:
`SDKROOT = macosx10.12;`
to
`SDKROOT = macosx;`
so that we won't need to touch each time the same file, when updating from your commits?
Many thanks!

igorkulman, 2020-07-04:
> Hi @zxystd can we kindly ask you to change your file **itlwm.xcodeproj/project.pbxproj** and replace inside the 3 instances of:
> 
> `SDKROOT = macosx10.12;`
> to
> `SDKROOT = macosx;`
> 
> ...so that we won't need to touch each time the same file, when updating from your commits?
> 
> I did that manually and it compiled great on Mojave 10.14.6 with Xcode 9.4.1 with the command provided in an earlier comment by @igorkulman above:
> 
> `xcodebuild -project itlwm.xcodeproj -target itlwm -configuration Release`
> 
> Remains to see if _that_ kext result works on my NUC! Many thanks.

I have an udpdate abash script that saves this change to git stash, pulls the latest code, applies the stash and builds the kext.

mackonsti, 2020-07-05:
Thanks @igorkulman but just modifying `SDKROOT` does not build a working kext on my Mojave with Xcode 9.4.1 in Terminal via the `xcodebuild` command. The resulting kext that is built without errors **cannot** get loaded, I get some error due to missing data in Info.plist (I see it's missing **CFBundlePackageType** key):
```
Defaulting to kernel file '/System/Library/Kernels/kernel'
Can't record kext in identifier lookup dictionary; no identifier.
Can't record kext in identifier lookup dictionary; no identifier.
```
Can you nevertheless point us to your bash script, please? Thanks...

Are you guys aware if Xcode > 9 is needed to successfully build this project? Is Mojave compatibility a pain to add?

igorkulman, 2020-07-05:
This is my complete `update.sh` script that builds the `kext` on Catalina using Xcode 11.5:

```bash
#!/bin/bash

# remove all local changes
git reset --hard HEAD
rm -rf build

# remove generated firmware
rm itlwm/FwBinary.cpp

# remove firmware for other wifi cards - DELETE OR CHANGE TO YOUR CARD
find itlwm/firmware/ -type f ! -name 'iwm-7260-*' -delete

# pull latest code
git pull

# add my networks - CHANGE TO YOUR NETWORKS
/usr/libexec/PlistBuddy itlwm/Info.plist -c "Set :IOKitPersonalities:itlwm:WiFiConfig:WiFi_1:ssid X"
/usr/libexec/PlistBuddy itlwm/Info.plist -c "Set :IOKitPersonalities:itlwm:WiFiConfig:WiFi_1:password XXX"

/usr/libexec/PlistBuddy itlwm/Info.plist -c "Set :IOKitPersonalities:itlwm:WiFiConfig:WiFi_2:ssid Y"
/usr/libexec/PlistBuddy itlwm/Info.plist -c "Set :IOKitPersonalities:itlwm:WiFiConfig:WiFi_2:password YYY"

/usr/libexec/PlistBuddy itlwm/Info.plist -c "Set :IOKitPersonalities:itlwm:WiFiConfig:WiFi_3:ssid Z"
/usr/libexec/PlistBuddy itlwm/Info.plist -c "Set :IOKitPersonalities:itlwm:WiFiConfig:WiFi_3:password ZZZ"

# remove unused network
/usr/libexec/PlistBuddy itlwm/Info.plist -c 'Delete :IOKitPersonalities:itlwm:WiFiConfig:WiFi_4'

# generate firmware
xcodebuild -project itlwm.xcodeproj -target fwgen -configuration Release -sdk macosx10.15

# build the kext
xcodebuild -project itlwm.xcodeproj -target itlwm -configuration Release -sdk macosx10.15
```

Just beware that I remove firmware that is not for my AC 7260 to reduce the size (2 MB vs 16 MB when built with everything)

xrokz, 2020-07-11:
Thanks Guys i will close it i forgot it open, 
seems that there is people got good help here

closed 2020-07-11 by xrokz

racka98, 2020-11-14:
> This is my complete `update.sh` script that builds the `kext` on Catalina using Xcode 11.5:
> 
> ```shell
> #!/bin/bash
> 
> # remove all local changes
> git reset --hard HEAD
> rm -rf build
> 
> # pull latest code
> git pull
> 
> # remove generated firmware
> rm itlwm/FwBinary.cpp
> 
> # remove firmware for other wifi cards - DELETE OR CHANGE TO YOUR CARD
> find itlwm/firmware/ -type f ! -name 'iwm-7260-*' -delete
> 
> # set version as commit count
> version=`git rev-list HEAD | wc -l | tr -d ' '`
> /usr/libexec/PlistBuddy itlwm/Info.plist -c "Set :CFBundleShortVersionString $version"
> /usr/libexec/PlistBuddy itlwm/Info.plist -c "Set :CFBundleVersion $version"
> 
> # add my networks - CHANGE TO YOUR NETWORKS
> /usr/libexec/PlistBuddy itlwm/Info.plist -c "Set :IOKitPersonalities:itlwm:WiFiConfig:WiFi_1:ssid X"
> /usr/libexec/PlistBuddy itlwm/Info.plist -c "Set :IOKitPersonalities:itlwm:WiFiConfig:WiFi_1:password XXX"
> 
> /usr/libexec/PlistBuddy itlwm/Info.plist -c "Set :IOKitPersonalities:itlwm:WiFiConfig:WiFi_2:ssid Y"
> /usr/libexec/PlistBuddy itlwm/Info.plist -c "Set :IOKitPersonalities:itlwm:WiFiConfig:WiFi_2:password YYY"
> 
> /usr/libexec/PlistBuddy itlwm/Info.plist -c "Set :IOKitPersonalities:itlwm:WiFiConfig:WiFi_3:ssid Z"
> /usr/libexec/PlistBuddy itlwm/Info.plist -c "Set :IOKitPersonalities:itlwm:WiFiConfig:WiFi_3:password ZZZ"
> 
> # remove unused network
> /usr/libexec/PlistBuddy itlwm/Info.plist -c 'Delete :IOKitPersonalities:itlwm:WiFiConfig:WiFi_4'
> 
> # generate firmware
> xcodebuild -project itlwm.xcodeproj -target fwgen -configuration Release -sdk macosx10.15
> 
> # build the kext
> xcodebuild -project itlwm.xcodeproj -target itlwm -configuration Release -sdk macosx10.15
> ```
> 
> Just beware that I remove firmware that is not for my AC 7260 to reduce the size (2 MB vs 16 MB when built with everything)

I've updated this script for anyone that comes across this as I did when I had the same issue. I've adapted all the changes that happened over time.

```shell
#!/bin/bash

# remove all local changes
git reset --hard HEAD
rm -rf build

# pull latest code
git pull

# remove generated firmware
rm include/FwBinary.cpp

# remove firmware for other wifi cards - DELETE OR CHANGE TO YOUR CARD
find itlwm/firmware/ -type f ! -name 'iwm-7265-*' -delete

# generate firmware
xcodebuild -project itlwm.xcodeproj -target fw_gen -configuration Release -sdk macosx

# build the kexts
## 1. itlwm.kext
xcodebuild -project itlwm.xcodeproj -target itlwm -configuration Release -sdk macosx

## 2. AirportItlwm Catalina
xcodebuild -project itlwm.xcodeproj -target AirportItlwm-Mojave -configuration Release -sdk macosx

## 3. AirportItlwm Catalina
xcodebuild -project itlwm.xcodeproj -target AirportItlwm-Catalina -configuration Release -sdk macosx

## 4. AirportItlwm Big Sur
xcodebuild -project itlwm.xcodeproj -target AirportItlwm-Big\ Sur -configuration Release -sdk macosx

# Location of Kexts
echo "You kexts are in build/Release!!"
echo " "
```

**Remember to change the wifi firmware part to yours or completely delete it to compile with all the firmwares.**

mjsorribas, 2021-03-12:
I tried some options on this thread but I only received this error:
```
itlwm/build/itlwm.build/Release/AirportItlwm-Catalina.build/DerivedSources/AirportItlwm_info.c:1:10: fatal error: 
      'mach/mach_types.h' file not found
#include <mach/mach_types.h>
         ^~~~~~~~~~~~~~~~~~~
1 error generated.

** BUILD FAILED **

The following build commands failed:
	ProcessPCH++ /itlwm/build/SharedPrecompiledHeaders/SharedPrecompiledHeaders/3736923090346969909/PrivateSPI.pch.gch /itlwm/itlwm/PrivateSPI.pch normal x86_64 c++ com.apple.compilers.llvm.clang.1_0.compiler
	CompileC /itlwm/build/itlwm.build/Release/AirportItlwm-Catalina.build/Objects-normal/x86_64/AirportItlwm_info.o /itlwm/build/itlwm.build/Release/AirportItlwm-Catalina.build/DerivedSources/AirportItlwm_info.c normal x86_64 c++ com.apple.compilers.llvm.clang.1_0.compiler
(2 failures)
```
somebody has the same error?