OpenIntelWireless/itlwm #159: Release and Big Sur Support

hierimpaug, 2020-07-02:
Hi! Thanks for all the hard work you've been putting in this project. If I may ask what percentage are you in now in the development progress and how far will it take more to push for a Release of a 1.0? Do you also have plans for macOS Big Sur? Thanks again!

williambj1, 2020-07-02:
No plans for Big Sur, since it's already perfectly supported.

[image]

ErrorErrorError, 2020-07-03:
Man that checkmark looks so misplaced on Big Sur lol. I will update WiFiMenuItemView with some improvements and fixed with wrapping text.

williambj1, 2020-07-03:
It seems that Big Sur's macOS SDK has a broken icon set implementation.

Icons appear correctly when I build HeliPort with Xcode 12 Beta and Catalina's SDK.

[image]

ErrorErrorError, 2020-07-03:
> It seems that Big Sur's macOS SDK has a broken icon set implementation.
> 
> Icons appear correctly when I build HeliPort with Xcode 12 Beta and Catalina's SDK.
> 
> [image]

I was actually talking about the position of the check in relation to the "Turn Wi-Fi Off" text but now that you mentioned I can see the different check icon appearance lol

closed 2020-07-08 by zxystd