OpenIntelWireless/itlwm #160: Connects to Wifi but does not get IP address after switching networks a few times with HeliPort

igorkulman, 2020-07-03:
On Intel 2760 AC when I just load the kext with my networks in Info.plist it always works fine.

With HelkiPort when I connected to a different network only most only works on second try, never on first try. Sometime I get to a state when I cannot connect back to my original network.

In my example, logs attached, I connected to K-guest and then was trying to connect back to K.

Although the status was connected I either got not IP address

[image]

Or a self assigned IP address

[image]

I tried renewing DHCP lease in System Preferences but it did not help.

I had to unload the kext and load it again to make it work again with K.

Logs: 
[itlwm.log](https://github.com/zxystd/itlwm/files/4871141/itlwm.log)

ErrorErrorError, 2020-07-04:
More information: I noticed after reading the state of the WiFi card while trying to authenticate the SSID, reguardless if the password is right or wrong, the status when it's trying to connect actually returns `ITL80211_S_SCAN` rather than `ITL80211_S_ASSOC` or `ITL80211_S_AUTH`.

zxystd, 2020-07-11:
can you check the latest code? thank you.

closed 2020-07-20 by igorkulman