OpenIntelWireless/itlwm #163: Requesting versioning and optimised 'Release' version in Git

mackonsti, 2020-07-04:
Hello @zxystd thank you for your continuous work on this driver!

Please, can I suggest that you include some form of versioning in your driver? Perhaps date-based inside the **Info.plist** ?

Both **CFBundleShortVersionString** and **CFBundleVersion** have not changed, kindly consider a release based on date e.g. 2020-07-04 that will always appear incremental...

Also, can I ask you why both _Release_ and _Debug_ compiled versions have the same size, 16MB is really a lot for a driver already... I was under the impression only "debug" contains symbols and DSYM and all that extra stuff.

Can you please check the project settings for Debug and Release and possibly optimise the result?

Not sure if it was asked already, can you expand more on the fact that the driver is 16 MB as one-big executable? Thanks!

williambj1, 2020-07-05:
Mojave is supported.

You can delete other firmware and leave the one your card uses then compile the kext, this will significantly reduce the kext’s size. No macros are used to reduce kext size on release builds so they’re the same.

Using date as version is not a proper way of versioning, GitHub releases will come soon once HeliPort’s bugs are fixed and itlwm’s  CI and CD gets implemented.

mackonsti, 2020-07-05:
Thank you for the tip and versioning news, appreciate it.

On a side note, please, how can we find out the detect / loaded firmware for the specific card? Is the exact firmware file name or model found in debug output of kernel? Today, I have no idea which file is used, to be honest. Thank you.

williambj1, 2020-07-05:
The firmware file’s name simply the same as your card model, located inside the firmware folder.

You don't have to dig into those fancy logs.

mackonsti, 2020-07-05:
Thank you @williambj1 however my Intel NUC is using 802.11ac wireless 9560 chipset [8086:9df0] hence the question :-) There is no such firmware file, so it must be something compatible with the existing files. Could it be **iwm-9260-34** ? Possibly, I am not sure. I would certainly like the debug output a little more verbose... The only readable thing I get is:
```
(itlwm) itlwm: iwm_match devId=0x9DF0
(itlwm) itlwm: itlwm: hw rev 0x310, fw ver 34.3125811985.0, address XYZ
```
Thanks!

williambj1, 2020-07-05:
9000

closed 2020-07-17 by zxystd