OpenIntelWireless/itlwm #164: When switching from "DHCP" to "Off" the device still shows Status: Connected

mackonsti, 2020-07-05:
Hello, I would like to report that with the code (built yesterday 4 July) the device does not seem "off" when switching from DHCP to OFF in the drop-down list of **Configure IPv4** as the light still remains _GREEN_ next to the controller name (left) and _STATUS_ still shows as connected:
[image]
[image]
Controller renamed to WLAN to distinguish from LAN; running on latest Mojave, not sure if this works in Catalina.

Output of **ifconfig** shows it as active, too (removed MAC addresses below) name is **en6**:
```
lo0: flags=8049<UP,LOOPBACK,RUNNING,MULTICAST> mtu 16384
	options=1203<RXCSUM,TXCSUM,TXSTATUS,SW_TIMESTAMP>
	inet 127.0.0.1 netmask 0xff000000 
	inet6 ::1 prefixlen 128 
	inet6 fe80::1%lo0 prefixlen 64 scopeid 0x1 
	nd6 options=201<PERFORMNUD,DAD>
gif0: flags=8010<POINTOPOINT,MULTICAST> mtu 1280
stf0: flags=0<> mtu 1280
XHC0: flags=0<> mtu 0
XHC20: flags=0<> mtu 0
en0: flags=8863<UP,BROADCAST,SMART,RUNNING,SIMPLEX,MULTICAST> mtu 1500
	options=b<RXCSUM,TXCSUM,VLAN_HWTAGGING>
	ether (-=REMOVED=-)
	inet6 fe80::1074:d9e6:1e00:e58d%en0 prefixlen 64 secured scopeid 0x6 
	inet6 fdc4:15e0:decc::100c:ac08:bb46:a0fb prefixlen 64 autoconf secured 
	inet6 fdc4:15e0:decc::a145:efcc:65d4:6fc6 prefixlen 64 autoconf temporary 
	inet 192.168.1.201 netmask 0xffffff00 broadcast 192.168.1.255
	inet6 fdc4:15e0:decc::201 prefixlen 64 dynamic 
	nd6 options=201<PERFORMNUD,DAD>
	media: autoselect (1000baseT <full-duplex,flow-control,energy-efficient-ethernet>)
	status: active
utun0: flags=8051<UP,POINTOPOINT,RUNNING,MULTICAST> mtu 2000
	inet6 fe80::64dd:3304:98e2:e2e6%utun0 prefixlen 64 scopeid 0x7 
	nd6 options=201<PERFORMNUD,DAD>
en6: flags=8863<UP,BROADCAST,SMART,RUNNING,SIMPLEX,MULTICAST> mtu 1482
	ether (-=REMOVED=-)
	inet6 fe80::2b:5fd6:adfc:a61a%en6 prefixlen 64 secured scopeid 0x8 
	inet6 fdc4:15e0:decc::148b:5a32:bd97:4b71 prefixlen 64 autoconf secured 
	inet6 fdc4:15e0:decc::34b0:a0b2:6e7b:ac5d prefixlen 64 autoconf temporary 
	nd6 options=201<PERFORMNUD,DAD>
	media: autoselect
	status: active
```
I think only IPv4 is off, whilst IPv6 as you can see remains assigned?
Thank you!

williambj1, 2020-07-05:
Because IPv6 is still on

closed 2020-07-05 by zxystd

mackonsti, 2020-07-05:
Hi and thanks, but how can I turn off IPv6 for this card-device, there is no such setting in System Preferences. I did check this before posting here. When I set from DHCP to OFF is it not supposed to make both IPv4 and IPv6 off?

williambj1, 2020-07-05:
So what’s your final goal actually? Just switching off DHCP?

If you’re trying to switch off the internet interface it’s better to use ifconfig (Interface BSD name) down.
If you’re trying to turn off your wifi card, you could do that in HeliPort

mackonsti, 2020-07-05:
Thank you @williambj1 I was not aware of the use of HeliPort, now I know. I will try compiling it. Thank you.