OpenIntelWireless/itlwm #165: your readme doesn't call out support for intel

myusrn, 2020-07-05:
Your english readme document [ https://github.com/OpenIntelWireless/itlwm/blob/master/.github/README_en.md ] doesn't list the "8 Series:8260, 8265" in supported devices section like the main chinese language readme document [ https://github.com/OpenIntelWireless/itlwm/blob/master/README.md ] does.  

I suggest updating it to match as otherwise folks might be mislead into thinking 8 series adapters are not supported.

zxystd, 2020-07-05:
thanks, the readme will be rewrite soon, add supported device ids instead of device product name.

myusrn, 2020-07-06:
Thanks for details. I'll let you close this issue when you see fit. Looking forward to testing this kext with intel ax200 / wifi6 m2 adapter.

mjsorribas, 2020-07-10:
It will be interesting if you share details of the install process after the compile process with xcode.

myusrn, 2020-07-10:
@mjsorribas are you raising a new english readme.md related issue or is this feedback for the main readme.md provided in the itlwm repo?

closed 2020-08-15 by williambj1