OpenIntelWireless/itlwm #166: [Bug] No wifi connection upon wake from sleep

naveenkrdy, 2020-07-06:
**Description**
Whenever i wake my laptop from sleep .
it shows that its connected in heliport but i get no internet.
then to fix it i have to turn off wifi and turn it back on through heliport.

**To Reproduce**
1. put the laptop to sleep 
2. wake it back up after few minutes
3. check internet connection in browser

**Expected behavior**
Should be connected upon wake from sleep.

**Environment (please complete the following information):**
 - OS Version: 10.15.5
- Software version : Using latest commit built from master as of now.
 - Hardware : Intel AX200
 - Laptop : DELL XPS 7590

- Thanks !

williambj1, 2020-07-06:
Logs please

naveenkrdy, 2020-07-07:
here you go.
[log.txt](https://github.com/OpenIntelWireless/itlwm/files/4884848/log.txt)

zxystd, 2020-07-07:
it will resume after a few minutes... but it is really a issue.

zxystd, 2020-07-07:
[itlwmx_0707.kext.zip](https://github.com/OpenIntelWireless/itlwm/files/4885240/itlwmx_0707.kext.zip)

I think I have fix it, please check.

naveenkrdy, 2020-07-07:
@zxystd that fixed the issue .
thanks alot ! amazing work !

closed 2020-07-07 by naveenkrdy