OpenIntelWireless/itlwm #168: Intel AC9560 (9462) of my HuaWei MateBook 13 is not working.

yuhr-zz, 2020-07-07:
First, thank you for your working at this project!

My machine is **HuaWei MateBook 13**, has the following network card:

---------------------------------------------------------------------------------
Intel AC9560 (9462)
VendorID 0x8086 DeviceID 0x02F0 SubVendorID 0x8086 SubDeviceID 0x2034
---------------------------------------------------------------------------------

It's not working!! Please tell me what's bad.

At command line,

_kextstat | grep -i itlwmx_

get the following result:
---------------------------------------------------------------------------------
   73    0 0xffffff7f8363a000 0x3a9000   0x3a9000   com.zxystd.itlwmx (1) 01BD5567-F512-34D8-8BF2-0D2025C57511 <18 13 6 5 3 1>
---------------------------------------------------------------------------------

_log show --last boot | grep itlwmx_

get the following result:
---------------------------------------------------------------------------------

2020-07-08 07:27:02.689823+0800 0xe6       Default     0x0                  0      0    kernel: (itlwmx) _start(struct kmod_info*, void*) has been invoked
2020-07-08 07:27:02.689935+0800 0xe6       Default     0x0                  0      0    kernel: (itlwmx) itlwm: free
2020-07-08 07:27:02.689984+0800 0xe7       Default     0x0                  0      0    kernel: (itlwmx) itlwm: free
2020-07-08 07:27:02.689986+0800 0xe8       Default     0x0                  0      0    kernel: (itlwmx) itlwm: free
2020-07-08 07:27:02.690756+0800 0xea       Default     0x0                  0      0    kernel: (itlwmx) itlwm: free
2020-07-08 07:27:02.690796+0800 0xec       Default     0x0                  0      0    kernel: (itlwmx) itlwm: free
2020-07-08 07:27:02.690797+0800 0xeb       Default     0x0                  0      0    kernel: (itlwmx) itlwm: free
2020-07-08 07:27:02.691407+0800 0xf6       Default     0x0                  0      0    kernel: (itlwmx) itlwm: free
2020-07-08 07:27:02.691418+0800 0xf9       Default     0x0                  0      0    kernel: (itlwmx) itlwm: free
2020-07-08 07:27:02.691462+0800 0xee       Default     0x0                  0      0    kernel: (itlwmx) itlwm: free
2020-07-08 07:27:02.691463+0800 0xf5       Default     0x0                  0      0    kernel: (itlwmx) itlwm: free
2020-07-08 07:27:02.691638+0800 0xfa       Default     0x0                  0      0    kernel: (itlwmx) itlwm: free
2020-07-08 07:27:02.692749+0800 0xf0       Default     0x0                  0      0    kernel: (itlwmx) itlwm: free
2020-07-08 07:27:02.692784+0800 0xf8       Default     0x0                  0      0    kernel: (itlwmx) itlwm: free
2020-07-08 07:27:02.694962+0800 0xf1       Default     0x0                  0      0    kernel: (itlwmx) itlwm: free
2020-07-08 07:27:02.694992+0800 0xf7       Default     0x0                  0      0    kernel: (itlwmx) itlwm: free
2020-07-08 07:27:02.695261+0800 0xef       Default     0x0                  0      0    kernel: (itlwmx) itlwm: free
2020-07-08 07:27:02.695274+0800 0xf4       Default     0x0                  0      0    kernel: (itlwmx) itlwm: free
2020-07-08 07:27:02.695455+0800 0xe9       Default     0x0                  0      0    kernel: (itlwmx) itlwm: free
2020-07-08 07:27:02.695574+0800 0xf2       Default     0x0                  0      0    kernel: (itlwmx) itlwm: free
2020-07-08 07:27:02.695581+0800 0xf3       Default     0x0                  0      0    kernel: (itlwmx) itlwm: free
2020-07-08 07:27:04.963994+0800 0x5dc      Default     0x0                  0      0    kernel: (itlwmx) itlwm: free
2020-07-08 07:27:04.966794+0800 0x5d8      Default     0x0                  0      0    kernel: (itlwmx) itlwm: free
2020-07-08 07:27:04.968368+0800 0x5d3      Default     0x0                  0      0    kernel: (itlwmx) itlwm: free
---------------------------------------------------------------------------------

foreveryyl, 2020-07-08:
hot reboot from windows

weidongbcn, 2020-07-08:
yuhr-zz，  you can read #139 .   how to do for working.

tctien342, 2020-07-08:
use `itlwm.kext` not X

Syedayy, 2020-07-08:
Using itlwm.kext does not fix the issue. 

I have a Huawei Matebook D15 with the exact same WiFi card (VendorID 0x8086 DeviceID 0x02F0 SubVendorID 0x8086 SubDeviceID 0x2034) and it doesn't connect to WiFi. My logs are also the same. 

I'm guessing this is because this is a gen-2 card, which is for itlwmx.kext but that doesn't work either.

lsvih, 2020-07-08:
Waiting for ax201 support..

zxystd, 2020-07-09:
02f0 is a gen2 device, I don't have this device to debug and test.
also, it is a duplicate issus, close.(#88 )

closed 2020-07-09 by zxystd