OpenIntelWireless/itlwm #170: AX200 Support

kodaman2, 2020-07-09:
How come this project says AX200 is supported but https://github.com/AppleIntelWifi/adapter says it will never support AX200, I thought your code was ported into that repo?

williambj1, 2020-07-09:
This project has nothing to do with other similar projects, AX200 is supported. Why not try it first before complaining ?

closed 2020-07-09 by williambj1

kodaman2, 2020-07-09:
First off no one is complaining, second lose the attitude. Is mainly curiosity, should have titled the issue better I am not complaining that AX200 is not working.

I'll ask on the other repo why they drop support for AX200, sorry for the inconvenience of asking a simple question.

This project or at least the first one by @zxystd was ported over the linked repo is clearly in the readme credits, so this project was tied to that one at one point, bud.

williambj1, 2020-07-09:
There’s no plan to backport, maybe you could contact it’s current maintainer. I didn’t meant to be aggressive, but only relied as fast as possible, sorry if that hurt you.

kodaman2, 2020-07-09:
Is all good man, I'll give this repo a try anyways and see how it goes. Just wanted to see my options for best wifi driver as this is my first opencore box. Thanks!

zxystd, 2020-07-09:
This is two different attempts on Intel WiFi. At present, hatf0 is mainly responsible for the other repo. He is trying to rewrite the entire 80211 stack. In my opinion, this will take a lot of time, and the code is relatively redundant and hard to debug for future. However, it is undeniable that it is a very promising project, but the workload is too heavy.

kodaman2, 2020-07-09:
Thanks for the info. I might get me a supported chip in the meantime and save the intel for a later build.

I've never done much driver programming, but if there's any documentation or links I'll like to learn a bit.