OpenIntelWireless/itlwm #171: 有办法在设置好的无线网后然后可以选着不同的无线网吗

Evan-tc, 2020-07-09:
就是设置好后台无线网名称和密码后，在界面上可以选着想要连接的无线网吗

rafhagunz, 2020-07-09:
仅在使用直升机场界面时才有可能 .

Evan-tc, 2020-07-09:
> 仅在使用直升机场界面时才有可能 .

什么是直升机场界面？ 我在itlwm把我两个无线网名称和密码都设置进去了，然后现在他会根据信号自动连接其中一个，我想能不能再无线网这边的界面里手动选择要连得那个网络，或者在网络设置里设置？有具体方法吗，麻烦了。

rafhagunz, 2020-07-09:
> > 仅在使用直升机场界面时才有可能 .
> 
> 什么是直升机场界面？ 我在itlwm把我两个无线网名称和密码都设置进去了，然后现在他会根据信号自动连接其中一个，我想能不能再无线网这边的界面里手动选择要连得那个网络，或者在网络设置里设置？有具体方法吗，麻烦了。

[HeliPort.app.zip](https://github.com/OpenIntelWireless/itlwm/files/4897156/HeliPort.app.zip)

安装并重新启动。记住该工具仍在开发中。

Evan-tc, 2020-07-09:
> > > 仅在使用直升机场界面时才有可能 .
> > 
> > 
> > 什么是直升机场界面？ 我在itlwm把我两个无线网名称和密码都设置进去了，然后现在他会根据信号自动连接其中一个，我想能不能再无线网这边的界面里手动选择要连得那个网络，或者在网络设置里设置？有具体方法吗，麻烦了。
> 
> [HeliPort.app.zip](https://github.com/OpenIntelWireless/itlwm/files/4897156/HeliPort.app.zip)
> 
> 安装并重新启动。记住该工具仍在开发中。

谢谢，只需要安装和重新启动吗，不用做额外的设置是吗

williambj1, 2020-07-09:
Use gitter for questions and personal support

closed 2020-07-09 by williambj1