OpenIntelWireless/itlwm #172: 网速监控下载速度一直是0

bininc, 2020-07-10:
[image]
如图上所示，使用软件是腾讯的Lemon

yuqZY, 2020-07-10:
是的，另外的外置的USB无线网卡也有这个问题，应该不是驱动的问题，可能跟系统有关系

zxystd, 2020-07-13:
确实有这个问题，但是在10.14以下的系统是正常的。这种问题优先级比较低，以后发现问题了再修复吧。

lang-ji, 2020-07-14:
10.15以后的很多系统都有这种问题，我的做法是照图中步骤把wifi硬件套一层桥接器，这样就能监听了
[image]
[image]

shomerchang, 2020-07-14:
嗯 是有这个问题

bininc, 2020-07-22:
> 10.15以后的很多系统都有这种问题，我的做法是照图中步骤把wifi硬件套一层桥接器，这样就能监听了
> [image]
> [image]

你这个方法看起来不错啊，试试看

sgzman, 2020-07-22:
使用PeakHour4这个软件试试

Misaka13514, 2020-08-04:
> 10.15以后的很多系统都有这种问题，我的做法是照图中步骤把wifi硬件套一层桥接器，这样就能监听了
> [image]
> [image]

成功解决
第一步是齿轮里管理虚拟接口
[创建方法](https://support.apple.com/zh-cn/guide/mac-help/mh43557/mac)

uifor, 2020-11-07:
macOS big sur 11.0.1下面在新建桥接器找不到Wi-Fi选项
[image]

llkhs, 2020-11-07:
[image]
我把wifi硬件套一层桥接器，似乎还是不能监听

llkhs, 2020-11-09:
> [image]
> 
> 我把wifi硬件套一层桥接器，似乎还是不能监听

请问是不是9560普遍都读不了下行网速？

lvjin250, 2020-11-14:
我也是这问题，AC9462，我还以为是我系统问题，重装了还是没解决，用网线就是正常的，WiFi就不行，虽然问题小，但逼死强迫症。上面说桥接WiFi的，我试了，苹果官网也说了，不能桥接无线网络，裂开了

Misaaya, 2020-11-15:
> 我也是这问题，AC9462，我还以为是我系统问题，重装了还是没解决，用网线就是正常的，WiFi就不行，虽然问题小，但逼死强迫症。上面说桥接WiFi的，我试了，苹果官网也说了，不能桥接无线网络，裂开了

最新的开发版本已经解决了这个问题，可以下载来试试

closed 2020-11-15 by zxystd