OpenIntelWireless/itlwm #173: Verbose output of the firmware applied for detected hardware

mackonsti, 2020-07-10:
Hi @zxystd thank you for your continuous effort in developing this driver.

Please, I would like to suggest and request to **verbosely show** the selected firmware loaded for a detected device in both Release and Debug versions. This would greatly help us identify which firmware file is used, because some of us have cards that work great but the card model (as name reference) is not applicable to the firmware names found in your project.

This is for compiled versions that are not "fat" but only using the needed firmware, in kext size :-)

For example, I use Intel® Wireless-AC 9560 in my NUC8i7BEH hack, and I was under the impression the driver loaded firmware **iwm-9260-34** but eventually [another member who can read the code, was kind to tell me](https://github.com/OpenIntelWireless/itlwm/issues/163#issuecomment-653915781) it's loading **iwm-9000-34** instead. So any verbose outfit of the applied firmware would be very welcome!

Thank you.

zxystd, 2020-07-13:
I will add firmware key info in logs soon, thanks for your advice.

zxystd, 2020-07-17:
link commit 4124360

closed 2020-07-17 by zxystd