OpenIntelWireless/itlwm #177: 一个奇怪的现象

XHL669, 2020-07-12:
就在今天下午出现了个奇怪的现象，应该也不是bug，可能是我哪里没搞明白，可以当听个故事吧，还请指教！

机型：灵越7590：AC9560

没接触这个驱动之前用的是comfast的USB wifi，电脑里安装着USB网卡的驱动。
然后我看这个项目不错，就在网上找了一个编译过的itlwm.kext，拖入并配置OC，搭配HeliPort的GUI使用。
然后我觉得运行一切正常就把USB网卡的驱动卸载了，之后一段时间没出现什么问题。
接下来的时间，我又鼓捣了一下其它的kext，把macos搞崩溃了。。。
进入Windows把EFI还原了一下，再进入macos，就出现了异常，任务栏wifi图标变成空心，再也无法打开wifi了。
接下来就是各种重启加还原，itlwm和itlwmx都试过，就是没有什么反应，清空kextcache也没用，一直无法通过HeliPort打开wifi。

最后我想：会不会是因为我卸载了原来的USB Wifi的驱动？
死马当活马医，我就把itlwm删除，然后清空了kextcache，然后重新安装了USB Wifi驱动。
再然后把itlwm.kext加回来。HeliPort居然就能打开wifi了！！！

所以。。。这个驱动能不能用还和Realtek的USB网卡发生关系了？？？？？
那是不是说我这个Realtek网卡驱动不能卸载了？。。。
这是怎么回事？

williambj1, 2020-07-12:
故事不错，但是与此驱动无关，可以重装 macOS 验证，就别折腾 kextcache 折磨自己了，引导工具注入他不香吗

closed 2020-07-12 by williambj1

XHL669, 2020-07-12:
哦？我在OCC里面添加了itlwm啊，依然不能通过HeliPort打开

williambj1, 2020-07-12:
考虑重装系统

XHL669, 2020-07-12:
可能和那次死机有关吧。。。
现在是正常的，不折腾了
但就是很奇怪，为什么USB驱动装回来就正常了。以前不正常的时候也用OCC配置过了，不应该是EFI的问题。

 

 
&mdash;

williambj1, 2020-07-12:
主要是 kextcache