OpenIntelWireless/itlwm #178: AC 9560 1600mhz says in windows, but not work with any itlwm or itlwmx

mrkaratechamp, 2020-07-13:
I have a huawei matebook 13 2020 (10 gen cpu i5),  everything works except wifi, it's suppose this wifi chip works with the current driver, but no matter what I tried, the kext loads ok I see numbers no erros with kextstat command, but no wifi, offcouse I double check my ssid and password before compile, also I test other builds from other people and change in plist file,  I'm using opencore and load them in plist config file, it's load ok.

I read there's a wifi cards with same model but are 2gen, could be this the reason?, will be suport for this 2gen cards?

thanks!

williambj1, 2020-07-13:
Logs please

mrkaratechamp, 2020-07-13:
here's the logs with itlwm:

[image]

williambj1, 2020-07-13:
We need full logs, not a screenshot

mrkaratechamp, 2020-07-13:
let me get them, but that two commands are ok? because that is not an extract, that's all the log, I put
kextstat | grep -i itlwm  
and 
log show --last boot | grep itlwm

do you need this ? or when you mean full logs should I use other command or get some log files from some folder?, 

thanks!

williambj1, 2020-07-13:
A simple google search would tell you how to dump console output into a file.

>> log.txt

mrkaratechamp, 2020-07-13:
yes, and if you don't want to help, don't reply me, what do you think you are?, I don't need your help if you will answer like that,
and look's like you don't know to read, I ask if you need more?, what extra log is necesary to dump, I don't ask how to dump the exact file I paste as image, why in the earth you will want that?, a simple search of how to be a person in google will help you.

zxystd, 2020-07-13:
According to the screenshot, I think your card id is `02f0`, this is gen2 device, I don't have this device before, now I have bought a new laptop with ax201, I am finding ways to support it, it seems not easy to support, more time needed for investigating. you can get the latest status in project tab. https://github.com/OpenIntelWireless/itlwm/projects/2
By the way, the most correct way of providing logs way is to attacht a log file. 
`log show --last boot | grep itlwm >~/Desktop/itlwm.log`
you can also get this way after some google search. Just Be patient, we works more other than with you, thanks.
Thought, the issue is already exists, please focus on this one #88 .

williambj1, 2020-07-13:
First of all, Github issues is a place to report bugs or request features but not for personal help request, if you really need help, you should really move to Gitter or forums.

Second, you did not provide any helpful information other than your card model, you didn’t even specify your card pci id, so are you expecting? Many people including me own a 9560ac and it works perfectly fine. Me and other devs don’t have time to waste on people like you that don’t even know how to write a proper bug report.

williambj1, 2020-07-13:
Duplicate of #88

closed 2020-07-13 by williambj1

mrkaratechamp, 2020-07-13:
@zxystd thanks, I also suspect it was a 2gen card and that's why others 9560 works but not mine, I will use your command for logs, and also any help for testing features  I can do and share it if helps, thanks!!.

@williambj1 please learn something from @zxystd , be person.

thanks!