OpenIntelWireless/itlwm #180: Slow speed!

barbarou, 2020-07-13:
Hi

When I do speed test I get about **30** mbps down and **12** mbps up on my 500/500 mbps plan. Any update/fix to get better speed? 

I use this kext [https://github.com/gnodipac886/MatebookXPro-hackintosh/tree/master/Intel-WiFi](url)

ErrorErrorError, 2020-07-13:
Speed is still a work in progress. It has not been fully implemented on either itlwm and itlwmx so slow speeds is what it should be expected, however you can try building the latest source close and see if it improves a bit.

barbarou, 2020-07-13:
> Speed is still a work in progress. It has not been fully implemented on either itlwm and itlwmx so slow speeds is what it should be expected, however you can try building the latest source close and see if it improves a bit.

I'm newbie, don't know how ti build one, and i can not find 8275 in supported list!

williambj1, 2020-07-14:
Performance issues and future plans will be documented later, unfortunately this would take quite a long time to improve. You can go to gitter for personal support, thanks.

closed 2020-07-14 by williambj1