OpenIntelWireless/itlwm #181: Performance improvements

groprima, 2020-07-13:
I would like to suggest some performance improvements to the project:

1. Implement correct log output depending on the build configuration

. Right now log is flooded with unrelated trace messages when release build is used. (see attachment)
2. Implement OSKextRequestResource to load firmware (see attachment). This will drastically reduce memory footprint, improve compile and kext loading speed. The build phase to copy firmware to the resource folder should be added to the project. The FWBinary.cpp becomes obsolete.
3. Implement correct device names (more cosmetic change, see attachment). 

The changes to the  itlwmx target are identical.

I'm not able to provide correct diff file because there are too many changes to the source on my side. You should be able to understand everything looking at attachment.

Thanks for your hard work.

[diff.txt](https://github.com/OpenIntelWireless/itlwm/files/4915482/diff.txt)

williambj1, 2020-07-14:
Can the firmware files load properly when it’s injected by a bootloader? This was a known issue before and this is why we have FWBinary.cpp

zxystd, 2020-07-14:
1.Now it is in WIP stage, need more logs, and I think the logs are necessary whenever it is release or debug mode. maybe change it in future(just a micro can control them).
2.`OSKextRequestResource` can not work in OC or CLOVER bootloader injection, should be install in L/E or S/L/E, this is not recommended.
3.Device name is not necessary for users or developers. they can see it in hackintool or system pci information and your implementation is wrong, they should be recognized not only by product ids.

thank you at all for your advice.

closed 2020-07-14 by zxystd

groprima, 2020-07-14:
You're right OSKextRequestResource won't work with boot loader. I think it makes sense to have a second optimized version to run from L/E. Hovewer, at this point, it is not a priority for you.