OpenIntelWireless/itlwm #183: 7260AC Not connecting.

hierimpaug, 2020-07-14:
Hi! I have been trying to make the itlwm.kext work but it unfortunately it won't work and it still show as Not Connected. As of now the kext is in C/K/O and I have tried to kextload but to no avail. 

[image]

Xpl0itU, 2020-07-14:
You have to specify the network details in the info.plist file and reboot

hierimpaug, 2020-07-14:
Yes I did specify the network details in the info.plist.

Xpl0itU, 2020-07-14:
unplug the USB WiFi adapter and reboot

hierimpaug, 2020-07-14:
@Xpl0itU I tried uninstalling the usb wifi adapter and rebuild the kext then load it/put in C/K/O still nothing. Tried to download Heliport and it shows the connected to the wifi icon on the menu bar but it is still not connected on the internet(Network Preferences)

zxystd, 2020-07-17:
check your EFI, this card actually can work.

closed 2020-07-17 by zxystd