OpenIntelWireless/itlwm #184: Intel 9560 0x02ED not working

ravnized, 2020-07-14:
Debug.log
`2020-07-14 03:39:18.124224-0700 0xde       Default     0x0                  0      0    kernel: (itlwm) _start(struct kmod_info*, void*) has been invoked
2020-07-14 03:39:18.128399-0700 0xdf       Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_match devId=0x9B41
2020-07-14 03:39:18.128435-0700 0xde       Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_match devId=0x9B61
2020-07-14 03:39:18.128474-0700 0xe1       Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_match devId=0x02F9
2020-07-14 03:39:18.128487-0700 0xe0       Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_match devId=0x1911
2020-07-14 03:39:18.128503-0700 0xe6       Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_match devId=0x02E0
2020-07-14 03:39:18.128537-0700 0xe7       Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_match devId=0x02C5
2020-07-14 03:39:18.128647-0700 0xe7       Default     0x0                  0      0    kernel: (itlwm) itlwm: free
2020-07-14 03:39:18.128647-0700 0xe1       Default     0x0                  0      0    kernel: (itlwm) itlwm: free
2020-07-14 03:39:18.128848-0700 0xe0       Default     0x0                  0      0    kernel: (itlwm) itlwm: free
2020-07-14 03:39:18.128848-0700 0xe6       Default     0x0                  0      0    kernel: (itlwm) itlwm: free
2020-07-14 03:39:18.129046-0700 0xde       Default     0x0                  0      0    kernel: (itlwm) itlwm: free
2020-07-14 03:39:18.129731-0700 0xe9       Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_match devId=0x02B4
2020-07-14 03:39:18.130818-0700 0xea       Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_match devId=0x0284
2020-07-14 03:39:18.131167-0700 0xeb       Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_match devId=0x02C8
2020-07-14 03:39:18.131476-0700 0xe3       Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_match devId=0x02EF
2020-07-14 03:39:18.132103-0700 0xe3       Default     0x0                  0      0    kernel: (itlwm) itlwm: free
2020-07-14 03:39:18.132107-0700 0xea       Default     0x0                  0      0    kernel: (itlwm) itlwm: free
2020-07-14 03:39:18.132108-0700 0xe9       Default     0x0                  0      0    kernel: (itlwm) itlwm: free
2020-07-14 03:39:18.132110-0700 0xdf       Default     0x0                  0      0    kernel: (itlwm) itlwm: free
2020-07-14 03:39:18.133554-0700 0xed       Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_match devId=0x02A4
2020-07-14 03:39:18.138331-0700 0xe2       Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_match devId=0x02ED
2020-07-14 03:39:18.139431-0700 0xeb       Default     0x0                  0      0    kernel: (itlwm) itlwm: free
2020-07-14 03:39:18.139556-0700 0xec       Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_match devId=0x02A3
2020-07-14 03:39:18.139589-0700 0xe5       Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_match devId=0x02E8
2020-07-14 03:39:18.141724-0700 0xf8       Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_match devId=0xF1A8
2020-07-14 03:39:18.141724-0700 0xe8       Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_match devId=0x02BC
2020-07-14 03:39:18.142543-0700 0xe8       Default     0x0                  0      0    kernel: (itlwm) itlwm: free
2020-07-14 03:39:18.142543-0700 0xec       Default     0x0                  0      0    kernel: (itlwm) itlwm: free
2020-07-14 03:39:18.142543-0700 0xe2       Default     0x0                  0      0    kernel: (itlwm) itlwm: free
2020-07-14 03:39:18.142902-0700 0xe5       Default     0x0                  0      0    kernel: (itlwm) itlwm: free
2020-07-14 03:39:18.153330-0700 0xf8       Default     0x0                  0      0    kernel: (itlwm) itlwm: free
2020-07-14 03:39:30.787799-0700 0x42a      Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_match devId=0x02A3
2020-07-14 03:39:30.791010-0700 0x42b      Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_match devId=0xA170
2020-07-14 03:39:30.794286-0700 0x427      Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_match devId=0x3E9B
2020-07-14 03:39:30.795140-0700 0x437      Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_match devId=0x02F0
2020-07-14 03:39:30.796862-0700 0x42b      Default     0x0                  0      0    kernel: (itlwm) itlwm: free
2020-07-14 03:39:30.796863-0700 0x437      Default     0x0                  0      0    kernel: (itlwm) itlwm: free
2020-07-14 03:39:30.796865-0700 0x427      Default     0x0                  0      0    kernel: (itlwm) itlwm: free
2020-07-14 03:39:30.796865-0700 0x42a      Default     0x0                  0      0    kernel: (itlwm) itlwm: free
2020-07-14 03:39:30.799277-0700 0x42a      Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_match devId=0x02A3
2020-07-14 03:39:30.806616-0700 0x42a      Default     0x0                  0      0    kernel: (itlwm) itlwm: free
2020-07-14 03:39:31.412192-0700 0x427      Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_match devId=0x3E9B
2020-07-14 03:39:31.418617-0700 0x427      Default     0x0                  0      0    kernel: (itlwm) itlwm: free
`

[image]

using itlwm.kext and heliport

zxystd, 2020-07-14:
I think your card id is 0x02F0, not 0x02ED. see this issue. #88

closed 2020-07-14 by zxystd

williambj1, 2020-07-14:
Duplicate of #88