OpenIntelWireless/itlwm #185: Wireless-AC 9462 0x02F0

ravnized, 2020-07-14:
using itlwmx.text detected but not working even through the plist

rafhagunz, 2020-07-14:
I recommend itlwm.kext for AC 9462.

ravnized, 2020-07-14:
switched to itlwm and still not working

rafhagunz, 2020-07-14:
attach  logs please

ravnized, 2020-07-14:
```
2020-07-14 15:27:56.671281+0200 0xde       Default     0x0                  0      0    kernel: (itlwm) _start(struct kmod_info*, void*) has been invoked
2020-07-14 15:27:56.671634+0200 0xe1       Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_match devId=0x02F9
2020-07-14 15:27:56.671639+0200 0xe1       Default     0x0                  0      0    kernel: (itlwm) itlwm: free
2020-07-14 15:27:56.671667+0200 0xdf       Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_match devId=0x9B41
2020-07-14 15:27:56.671670+0200 0xdf       Default     0x0                  0      0    kernel: (itlwm) itlwm: free
2020-07-14 15:27:56.671975+0200 0xde       Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_match devId=0x9B61
2020-07-14 15:27:56.671976+0200 0xe0       Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_match devId=0x1911
2020-07-14 15:27:56.671983+0200 0xe0       Default     0x0                  0      0    kernel: (itlwm) itlwm: free
2020-07-14 15:27:56.671984+0200 0xde       Default     0x0                  0      0    kernel: (itlwm) itlwm: free
2020-07-14 15:27:56.672022+0200 0xe7       Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_match devId=0x02C5
2020-07-14 15:27:56.672026+0200 0xe7       Default     0x0                  0      0    kernel: (itlwm) itlwm: free
2020-07-14 15:27:56.672429+0200 0xe6       Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_match devId=0x02E0
2020-07-14 15:27:56.672434+0200 0xe6       Default     0x0                  0      0    kernel: (itlwm) itlwm: free
2020-07-14 15:27:56.672870+0200 0xec       Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_match devId=0x02A3
2020-07-14 15:27:56.672874+0200 0xec       Default     0x0                  0      0    kernel: (itlwm) itlwm: free
2020-07-14 15:27:56.676460+0200 0xe3       Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_match devId=0x02EF
2020-07-14 15:27:56.676464+0200 0xe3       Default     0x0                  0      0    kernel: (itlwm) itlwm: free
2020-07-14 15:27:56.676752+0200 0xe5       Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_match devId=0x02E8
2020-07-14 15:27:56.676756+0200 0xe5       Default     0x0                  0      0    kernel: (itlwm) itlwm: free
2020-07-14 15:27:56.678637+0200 0xea       Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_match devId=0x0284
2020-07-14 15:27:56.678641+0200 0xea       Default     0x0                  0      0    kernel: (itlwm) itlwm: free
2020-07-14 15:27:56.679432+0200 0xeb       Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_match devId=0x02C8
2020-07-14 15:27:56.679454+0200 0xeb       Default     0x0                  0      0    kernel: (itlwm) itlwm: free
2020-07-14 15:27:56.679602+0200 0xe9       Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_match devId=0x02B4
2020-07-14 15:27:56.679611+0200 0xe9       Default     0x0                  0      0    kernel: (itlwm) itlwm: free
2020-07-14 15:27:56.682829+0200 0xe4       Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_match devId=0x02F0
2020-07-14 15:27:56.682834+0200 0xe4       Default     0x0                  0      0    kernel: (itlwm) itlwm: free
2020-07-14 15:28:10.864977+0200 0x42a      Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_match devId=0x02A3
2020-07-14 15:28:10.864980+0200 0x42a      Default     0x0                  0      0    kernel: (itlwm) itlwm: free
2020-07-14 15:28:10.868489+0200 0x42b      Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_match devId=0xA170
2020-07-14 15:28:10.868493+0200 0x42b      Default     0x0                  0      0    kernel: (itlwm) itlwm: free
2020-07-14 15:28:10.868605+0200 0x425      Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_match devId=0x3E9B
2020-07-14 15:28:10.868608+0200 0x425      Default     0x0                  0      0    kernel: (itlwm) itlwm: free

```

ravnized, 2020-07-14:
[image]

lsvih, 2020-07-14:
Waiting for #127

hierimpaug, 2020-07-15:
@ravnized what tool is that? I'm also having the same problem with my 7260AC

ravnized, 2020-07-15:
hackintool. (Github)[https://github.com/headkaze/Hackintool](url)

zxystd, 2020-07-17:
link issues #127 #88

closed 2020-07-17 by zxystd

williambj1, 2020-07-17:
Duplicate of #88