OpenIntelWireless/itlwm #186: Connecting to a network that requires username and password

666ares, 2020-07-14:
How to connect to a network that requires username and password to connect to it?

techies23, 2020-07-15:
**This does not seem to be an issue you are asking in this issue thread.** In order for a developer or anyone to understand you need to clarify what you did and how you did to reach certain situation.

Anyway, if it helps - I am very new to Hackintosh however, i successfully implemented this awesome tool by the developer and it works pretty darn well i must say. Really a big thanks to the developer here !

I have a Lenovo y7000P 144hz series which had a RTL8822BE card which does not seem to work. So, switched it to Intel 9520 family since i could not find the Broadcom card on my side of the world. 

To enable it, just added **itlwm.kext** inside **EFI/Clover/kexts/other** and Installed "**HeliPort**" App, Restart and it seemed to do the trick. Although, i need to restart a few times to get the Wifi working sometimes but who cares i got it working for now.

Just ran a **sudo kextload -i /** thinking this would reset any cache ( please anyone let me know what exactly this does if you want )

This is all using Clover Configurator.

williambj1, 2020-07-15:
Enterprise-level encryption is currently not supported, stay tuned.

@techies23 This issue is actually valid, thank you for your interest in this project.

williambj1, 2020-07-15:
Duplicate of #161

closed 2020-07-15 by williambj1