OpenIntelWireless/itlwm #188: Awesome Work![Feature Request]

abaza7, 2020-07-15:
First of all, Thanks so much for your great efforts it works flawlessly for me (although a bit slower at 40mbps but I am not complaining lol). Do you have a paypal or a patreon if someone wants to support this awesome project?

Secondly, The feature request:
Would it be possible to inject it as an actual wifi instead of the ethernet spoofing? I have no idea how easy or hard it would be so I'm just proposing the idea. The thing is, if that is implemented that would suddenly mean that we can use handoff and continuity features without spending 80$+ on Wi-Fi cards.

Again thanks man!

ErrorErrorError, 2020-07-15:
> First of all, Thanks so much for your great efforts it works flawlessly for me (although a bit slower at 40mbps but I am not complaining lol). Do you have a paypal or a patreon if someone wants to [...]

From what I've read implementing the Apple802.11 stack is fairly complicated/time consuming and worst part is that is closed source. There was actually another repo that attempted to add native support but we'll I haven't heard anything from them in months. Maybe in the future it will be considered by the team once the intel cards are more stable, however don't quote me on that.

zxystd, 2020-07-15:
Just like above said, Apple80211 is closed source, no document, no sdks, system version dependent. We need too much time on reversing it and investigating how to use, and it will cause too much compatibility and stability problems. It is nonsense until we full implemented Intel card hardware features/net80211 stack(we SHOULD self implemented 80211 stack whether the Apple80211 is used).
BTW, airdrop/handoff/continuity  are entire large sub systems, different with you think there is a switch can enable it.
Thanks.

closed 2020-07-15 by zxystd

igorkulman, 2020-07-15:
@zxystd you should probably rename this issue to something like `[Feature request] Airdrop / Handoff / Continuity support` to make it stand out as a pinned issue.

zxystd, 2020-07-15:
> @zxystd you should probably rename this issue to something like `[Feature request] Airdrop / Handoff / Continuity support` to make it stand out as a pinned issue.

Yes, you are right.

Stevemoretz, 2020-08-16:
What about Hotspot?Is that impossible too?

ErrorErrorError, 2020-08-16:
> What about Hotspot?Is that impossible too?

Hotspot is unsupported as of now.

cleihen, 2020-09-08:
能给个内测群吗？

williambj1, 2020-09-17:
https://github.com/OpenIntelWireless/itlwm/tree/airport_impl