OpenIntelWireless/itlwm #189: [Bug]Cannot turn on WIFI after a shutdown from windows

XHL669, 2020-07-15:
**Describe the bug**
Cannot turn on WIFI after a shutdown from windows. I don't know if it is because of itlwm or the HeliPort GUI. It needs a forced shutdown to work properly.

**To Reproduce**
Steps to reproduce the behavior:
1. Shut dowm from Windows
2. Boot and Login in  macOS
3. The WIFI icon will turn into blank one and HeliPort cannot turn on WIFI. [image]

**Expected behavior**
Be able to turn on WIFI after a normal shutdown from Windows OS rather than a forced shutdown.

**Screenshots**
This is a screenshot after a normal shutdown from windows.
[image]
HeliPort cannot find the driver. Click "Turn on WIFI" but there is no response.

This is a screenshot after a forced shutdown from windows.
[image]

**Environment (please complete the following information):**
 - OS Version: macOS 10.15.5

williambj1, 2020-07-15:
Duplication of #139

closed 2020-07-15 by williambj1

XHL669, 2020-07-15:
> Duplicate of #139

I see, but how to cold boot except a forced shutdown?
I have already disabled "Quick boot" in Windows OS and "Secure Boot" in BIOS. What's more?

ErrorErrorError, 2020-07-15:
> > Duplicate of #139
> 
> I see, but how to cold boot except a forced shutdown?
> I have already disabled "Quick boot" in Windows OS and "Secure Boot" in BIOS. What's more?

Still a WIP. As of now the only way to fix is to cold boot, i.e. forced shutdown.

williambj1, 2020-07-15:
Really not that convenient, but for now that's the only solution, unfortunately.

XHL669, 2020-07-15:
> > > Duplicate of #139
> > 
> > 
> > I see, but how to cold boot except a forced shutdown?
> > I have already disabled "Quick boot" in Windows OS and "Secure Boot" in BIOS. What's more?
> 
> Still a WIP. As of now the only way to fix is to cold boot, i.e. forced shutdown.

I am worried that long pressing power key to shut down could damage the computer such as the SSD, but now it is the only way.
Anyway, thank you both.

zxystd, 2020-07-15:
> I am worried that long pressing power key to shut down could damage the computer such as the SSD, but now it is the only way.
> Anyway, thank you both.

You don't need to force shutdown, shutdown only as well. don't direct reboot to Mac, that will not have problem.

XHL669, 2020-07-15:
> > > > > Duplicate of #139
> > > > 
> > > > 
> > > > I see, but how to cold boot except a forced shutdown?
> > > > I have already disabled "Quick boot" in Windows OS and "Secure Boot" in BIOS. What's more?
> > > 
> > > 
> > > Still a WIP. As of now the only way to fix is to cold boot, i.e. forced shutdown.
> > 
> > 
> > I am worried that long pressing power key to shut down could damage the computer such as the SSD, but now it is the only way.
> > Anyway, thank you both.
> 
> You don't need to force shutdown, shutdown normally as well. don't direct reboot to Mac, that will not have problem.

OK, thank you. Right now @williambj1 told me that I need shutdown without charging.